Snapmaker/Snapmaker2-Controller #5: Feature request: working time

ITmaze, 2020-11-19:
Is it possible to add on the info page, or in a new maintenance page, of the touchscreen the total working hours of the machine? I think it will be very useful if we could have the n. of the hours the machine worked (printing, CNC, laser…) so that we can see if some maintenance is needed or not.
Also a reset button will be fine so that we could reset after maintenance or cleaning up or…

What do you think?

Reference: @Airmonkdev
Source: https://github.com/ITmaze/snapmaker-firmware/issues/1

ITmaze, 2020-11-24:
This issue was transferred to the official repository.

Source: https://github.com/Snapmaker/Snapmaker2-Controller/issues/5

ITmaze, 2020-11-24:
Gotta love technology, closed the issue in the wrong repository. Sorry for the noise.

scotthsl, 2020-12-08:
> Gotta love technology, closed the issue in the wrong repository. Sorry for the noise.

Never mind, our production manager knew your needs. 

For now, there are two working modes in Snapmaker 2.0: offline working with Screen and inline working with PC.

We can record the offline working hours conveniently, because Screen have recorded the operations of START/PAUSE/RESUME/STOP/FINISH, so it can simply figure out the duration of working.

But for inline working, it's not easy to accurately record the working time by controller, for example, someone just turn on the CNC by Gcode **M3 P100**, then power off the machine soon after.

ITmaze, 2020-12-09:
Did you see this PR on the Marlin issue tracker?
- https://github.com/MarlinFirmware/Marlin/pull/3625

There is also this code for Ultimaker2 which writes to EEPROM (expected life 11 years, see comments.)
- https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/lifetime_stats.cpp