Snapmaker/Snapmaker2-Controller #6: Implement M600 - Filament Change

ITmaze, 2020-11-19:
SM2 will stop on M600, but it won't resume because this function has not been implemented.

Reference: @ITmaze
Source: https://github.com/ITmaze/snapmaker-firmware/issues/2

scotthsl, 2020-11-20:
We implemented another version of PAUSE/RESUME functions for SM2, so it doesn't support to be resumed by Gcode.

ITmaze, 2020-11-20:
Can this be mapped to the GCODE, so filament changes can be accomplished
inline?

scotthsl, 2020-11-29:
For now, it can be remapped to _filament runout_ event by calling **system.PauseTrigger(TRIGGER_SOURCE_RUNOUT)** in **GcodeSuite::M600()** , then APP on Screen also get the event and pop the panel for filament change.

odrobnik, 2020-12-21:
Yes, I think that it would be awesome to support manual filament change after certain layers in Luban!

Bsherba, 2021-01-11:
So the snapmaker will extrude about 10mm of filament after a reloaded runout, which you can intercept with your hand real quick before it reaches your part again. this appears to get the flow moving nice and well.

Their starter gcodes always have it extrude material in the corner in the same fashion and wipe it off, they should probably be doing that in this scenario too.

Furthermore, defining the coordinates of the wipe would be ideal... for people like me that have clips to hold glass on

This is something that can be done in the start gcode, but it would be cool to implement a custom position for wiping the snot on your sleeve in a convenient position for your build after a filament runout resume.

odrobnik, 2021-03-30:
@scotthsl could you give some details how your workaround would be implemented. Is that something I could do on my machine, or would that have to go into the firmware?

xhycode, 2021-03-31:
Please refer to PR[#91](https://github.com/Snapmaker/Snapmaker2-Controller/pull/93) for your concern,  recently we will release a new version firmware, but it won't contain this feature. You will have the official support in next release.
Best regards
xhy

closed 2021-08-13 by xhycode