Snapmaker/Snapmaker2-Controller #9: Cannot disable door detection

flynny75, 2020-11-21:
https://github.com/Snapmaker/Snapmaker2-Controller/blob/122d90d0a25db256b1a3eead63d6212f74e84ff7/snapmaker/src/module/enclosure.cpp#L182

The enabled flag is never assigned. The default value is true, thus the door event is always processed and door detection cannot be disabled. I suspect it is meant to be assigned in the Enable and Disable methods.

scotthsl, 2020-11-29:
Thanks a lot , will check and fix it!

closed 2021-01-25 by scotthsl

ITmaze, 2021-01-26:
When you close an issue it's customary to link the commit that fixed it so
that if there's subsequently a problem, the issue can be reopened.

This makes for a much better way of tracking the relationship between the
code and the issue.