Snapmaker/Snapmaker2-Controller #10: Feature Request: Allow enclosure lighting to remain on when tool head is unpowered

OrionPirate, 2020-11-23:
Not certain this is firmware rather than hardware, but if it is firmware:

Disconnecting a tool head from the power bus causes the enclosure LED lighting to turn off, and it cannot be turned on until a tool head is attached to power. 

This makes changing tool head inside the enclosure MUCH more awkward, as lighting would be incredibly helpful in this specific case!

scotthsl, 2020-11-29:
Want to confirm firstly you unplugged your tool head while keeping the controller power on?

OrionPirate, 2020-11-29:
correct. 

snapmaker was powered on, enclosure lights were on. I unplugged the cable from the tool head, and the enclosure lights went out. Controller was still connected to the snapmaker at the time.

scotthsl, 2020-11-29:
> correct.
> 
> snapmaker was powered on, enclosure lights were on. I unplugged the cable from the tool head, and the enclosure lights went out. Controller was still connected to the snapmaker at the time.

Hi OrionPirate,

Got your needs, will try to improve the operations to change tool head.

 But please do not plug / unplug tool heads or any other modules again while controller is powered on.

Sorry for didn't make it clear in the guide book, and our colleague is improving it.

Thanks,
Scott

scotthsl, 2020-11-29:
Hot-plug may damage the circuit between Controller and modules, so please turn off the power firstly before plugging/unplugging modules.

PR #47 (2021-01-07): Enable Power Control
This pull request adds a new gcode command, M2001, to enable control of the power domains via gcode.

The motivation for this change is to enable safely changing toolheads while the enclosure lights stay on.

Resolves #10

Command Syntax:
M2001 S{0} T{0} B{0}
S: Screen power domain
T: Toolhead / linear module power domain
B: Bed / addon power domain

Example usage of this command:
M2001 T0 ; Disable toolhead / linear module power
M2001 T ; Enable toolhead / linear module power, automatically reboots the screen and main controller
M2001 S T B ; Enable all domains, rebooting [... cut]
commit 6a233889 (2021-01-07): Feature: Add M2001 gcode for power control

Command Syntax:
M2001 S{0} T{0} B{0}
S: Screen power domain
T: Toolhead / linear module power domain
B: Bed / addon power domain

Resolves #10
closed 2021-01-07 by scotthsl