Snapmaker/Snapmaker2-Controller #13: Feature Request: Enable M600 Filament Change

parachvte, 2020-12-01:
This is a duplicated issue in Luban repo (Snapmaker/Luban#248) from @tonejaco.

> For doing multi-color prints and other such custom things it would be good to have the "Pause" feature of the Touch Screen accessible thru a GCODE call. M600? Perhaps all that is needed is the correct syntax for executing it.

parachvte, 2020-12-01:
Also it's the same feature request as mentioned in https://github.com/Snapmaker/Snapmaker2-Controller/issues/6

shauncrain, 2020-12-21:
This is my number 1 need at the moment.  Fully support this request.

odrobnik, 2020-12-27:
👍

Bsherba, 2021-01-11:
So the snapmaker will extrude about 10mm of filament after a reloaded runout, which you can intercept with your hand real quick before it reaches your part again. this appears to get the flow moving nice and well.

Their starter gcodes always have it extrude material in the corner in the same fashion and wipe it off, they should probably be doing that in this scenario too.

Furthermore, defining the coordinates of the wipe would be ideal... for people like me that have clips to hold glass on

This is something that can be done in the start gcode, but it would be cool to implement a custom position for wiping the snot on your sleeve in a convenient position for your build after a filament runout resume.

closed 2021-08-13 by xhycode