Snapmaker/Snapmaker2-Controller #19: Feature Request: Stepper motor disengage after period of time

parachvte, 2020-12-01:
Issue from https://github.com/Snapmaker/Luban/issues/484

> Please consider a process improvement to cycle step motors down (turn off) after X amount of time (20 minutes?). Require a re-home before another action in case minor movements have occurred. This will help increase the longevity of the steppers.

FKX747, 2020-12-12:
Might be an easy one to fix. For me it looks like it is related to the following two sections in the Marlin Configuration_adv.h

https://github.com/Snapmaker/Snapmaker2-Controller/blob/a0e8e2c7f8e123cee3fe7aec35016393e8d2637d/Marlin/Configuration_adv.h#L788

https://github.com/Snapmaker/Snapmaker2-Controller/blob/a0e8e2c7f8e123cee3fe7aec35016393e8d2637d/Marlin/Configuration_adv.h#L526-L529

brent113, 2020-12-13:
It's not SD_FINISHED_...., it's this
```
/**
 * Idle Stepper Shutdown
 * Set DISABLE_INACTIVE_? 'true' to shut down axis steppers after an idle period.
 * The Deactive Time can be overridden with M18 and M84. Set to 0 for No Timeout.
 */
#define DEFAULT_STEPPER_DEACTIVE_TIME 120
#define DISABLE_INACTIVE_X false
#define DISABLE_INACTIVE_Y false
#define DISABLE_INACTIVE_Z false // Set 'false' if the nozzle could fall onto your printed part!
#define DISABLE_INACTIVE_E false
```
Change 120s to 1200 for 20min, then all steppers get flagged as true.