Snapmaker/Snapmaker2-Controller #20: Feature Request: 3D Print load/unload filament on paused state

Airmonkdev, 2020-12-04:
I saw we can pause a 3D Print job from the touch screen and resume later.
When the job is in paused state the module is in home position and we can access and modify some parameters from the touch screen (bed temperature, nozzle temperature ....).
Would it be possible to add the possibility to also change the filament in paused state with the two button ''load/unload"?
This feature will let us change manually the filament color of the printing object just manually pausing the job at the wanted point.

scotthsl, 2020-12-08:
For now you can open the latch in 3DP tool head, then unload / load filament manually:relaxed:

yhira1023, 2020-12-08:
I tried  filament change on Post processing scripts by Cura ,could unloaded filament.
but new filament couldn't loaded because of not accept "continue operation"
If it cloud be ,can use multi filaments.and So happy!
plz consider this feature.
Thank you