Snapmaker/Snapmaker2-Controller #21: Feature Request: Allow Trusted Computers to Connect To WiFi Without Verification

Johnny132, 2020-12-07:
Having the Snapmaker 2 being able to connect to our computers through WiFi is very nice, but it turns into a pain for people who’s computer are a ways away from their printer. I feel there should be a trusted computer feature so you still have the security of verification on outside computer, but for you, you can access it much easier.

ruckc, 2020-12-07:
This would make my experience with Snapmaker so much better.

halr9000, 2020-12-07:
Please add this, my printer and PC are not close, so it would be a huge quality of life improvement to be able to trust my computer once.

ITmaze, 2020-12-07:
To notify the developers of this feature request, I'd recommend that you
create a GitHub issue here:
https://github.com/Snapmaker/Snapmaker2-Controller

ruckc, 2020-12-08:
@ITmaze that is where this issue is...

ITmaze, 2020-12-08:
Doh,

Sorry for the noise. I'm getting the forum via email and github
notifications, both sent to the same mailbox ;-)

Carry on ...

Mikul00, 2021-01-08:
I too really think needing the authorization limits my ability to print remotely.

PoetikDragon, 2021-01-08:
This really needs to be a thing.

NomadicVoxel, 2021-01-09:
This is a huge problem, agreed. 

I think there's an alternative idea though, what if instead of manually going down and hitting the button, you set up a password on the printer and have to input that in Luban to connect?

PatrickBig, 2021-01-10:
I would really like this to be a priority. Not a fan of two trips to the printer every time I set something up. Once to authorize and one to click "start job"

jenkins007, 2021-01-25:
Please fix this. After the snapmaker is turned off, then back on, a screen request is required. My printer is 3 rooms away and it is very frustrating to have to walk to it every day I want to print.

ruckc, 2021-01-25:
@jenkins007 - my workaround due to this was to install octoprint on a raspberry pi, and use it instead of luban... don't believe it supports the laser or cnc modules though.

jafischer, 2021-01-27:
Just another vote for this feature. My printer is only in the next room and it still drives me crazy to have to make the trek back and forth multiple times.

You don't have to worry about making it 100% secure and fool-proof. Just make it opt-in, so we can make the choice. My WiFi is secure, so it's not like a neighbor can start a print job or anything...

Fanfwe, 2021-02-17:
I'd like to see this one too. Octoprint on a Raspberry Pi is an okay workaround, but you'd loose features like resuming after power failure, for example. When the printer is in a trusted network, we should be able to set an option that bypasses the verification and just accepts connection from Luban right away. That option would be disabled by default, but anybody who willingly wants to lower the level of security should be able to do so.

pegak, 2021-03-02:
I vote for this feature. Ideally have an API for for it, so it can be used even from other tools with plugins (Cura).

parachvte, 2021-03-03:
> I vote for this feature. Ideally have an API for for it, so it can be used even from other tools with plugins (Cura).

There is already a [Cura plugin](https://github.com/macdylan/Snapmaker2Plugin ) for Snapmaker 2.0 from Snapmaker user, though its document is all in Chinese. It connects to SM 2.0 through Wi-Fi, and remember the authentication token, which allows you to verify only once.

NomadicVoxel, 2021-03-03:
> There is already a [Cura plugin](https://github.com/macdylan/Snapmaker2Plugin) for Snapmaker 2.0 from Snapmaker user, though its document is all in Chinese. **It connects to SM 2.0 through Wi-Fi, and remember the authentication token, which allows you to verify only once.**

Hold on a second. Recallable authentication tokens are already a feature. And Luban doesn't use it. *What?*

parachvte, 2021-03-04:
Checkout latest [firmware log](https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/12), the improved firmware was released last month. While Luban haven't release its update yet.

scyto, 2021-07-21:
> There is already a [Cura plugin](https://github.com/macdylan/Snapmaker2Plugin) for Snapmaker 2.0 from Snapmaker user, though its document is all in Chinese. It connects to SM 2.0 through Wi-Fi, and [...]

I translated this text and awaiting cura maintainer to approve the PR.

Eddie-Hartman, 2022-02-05:
@Johnny132 I believe your original issue has been resolved if you upgrade to latest firmware. Luban gets a token that remembers you if you try to connect to the machine again so you can connect without verification.