Snapmaker/Snapmaker2-Controller #23: Feature Request: Display MAC address on Snapmaker2.0 touch screen controller

blockmodule, 2020-12-09:
Feature Request: Display MAC address on Snapmaker2.0 touch screen controller

In my home, all of the WI-FI wireless LAN environments have MAC address restrictions in place, and you can't connect to the LAN without registering your MAC address.

However, the current Snapmaker 2.0 touch screen does not display the MAC address, and there is no way to fix this except to figure it out via the wireless LAN router.

In some cases, I believe that some models cannot even be referenced from the wireless LAN router.

We would like to suggest this as a feature request, as schools and companies often have MAC address connection restrictions in place.

voidrin, 2021-01-06:
Sorry for replying late, we have received this feature request few weeks ago, but at that time we still busy developing new modules for Snapmaker2 and we temporarily marked it as low priority.

We will keep following this and update status here, please stay tune.

ywabiko, 2021-01-25:
+1 for this feature. Another bug (cannot forget Wifi password) is making situation worse in MAC address filtering.

Scenario: To examine MAC address, you have to connect Snapmaker to guest Wifi once. From router management UI, you look up the last connected client that is Snapmaker. Now you add its MAC address to the white list and reconnect to the main Wifi network.
Now Snapmaker often connects to guest Wifi, because it cannot forget Wifi (even after factory reset).

Please implement one or both. If Snapmaker can show MAC address in the first place, this problem doesn't happen.