Snapmaker/Snapmaker2-Controller #25: Bug: Laser Auto Origin Incorrect Sometimes

brent113, 2020-12-11:
Occassionally when running multiple jobs, or cancelling a job and starting a new one, the Auto material thickness will set the origin to the wrong location.

The correct calculated initial work origin is the middle of the bed at the laser focal length. I have had incorrect locations include the maximum Z as the Z height, X max, a Z=97 once, and a few of other incorrect locations.

I found a workaround. When starting a job, go to manual, save origin, then back to auto and continue, it will recalculate the correct work origin.

scotthsl, 2020-12-12:
Thank you for your feedback, the Auto material thickness is implemented by the App running on screen, and this issue has been sent back to the colleague who is responsible for App development.
BTW, could you please tell us your firmware version?

brent113, 2020-12-12:
Firmware version is 1.10.1.

scotthsl, 2020-12-12:
Got it, thanks again, we will try to reproduce the bug.

brent113, 2020-12-12:
Most consistently reproduces after cancelling a job.

The particular file I am running runs G91 relative gcode. That may be important.

There are other bugs if the team is willing to look into additional issues with G91. The door detection and resume feature resets the coordinate space to G90, ruining jobs if the door is opened in the middle. Additionally run boundary incorrectly calculates for G91 relative gcode.

scotthsl, 2020-12-12:
Thank you very much, we are certainly willing to fix these issues.
We've been busy developing several new modules for snapmker 2.0 over the past few months, so no new firmware has been released. But the information you mention above are very important, it will allow us to reproduce quickly these issues and fix them.