Snapmaker/Snapmaker2-Controller #27: Start automatic calibration above the bed in stead of right behind it (move X &Y before down Z)

brvdboss, 2020-12-13:
When you start the calibration routine, the z-axis moves down first until a level that is expected to be a bit above the printbed.  The bed sensor however is located just behind the the bed and thus can not detect it.

If you are using another type of printbed (let's say you have put it on top of the cnc wasteboard) then it could be too thick and the print head smashes in to the print bed.  When the something is detected below the bed sensor it halts as expected.
See the location on start of callibration:
[image]

I would suggest to move towards the first position by moving X&Y first and only then Z.  That would prevent this problem completely.

scotthsl, 2020-12-25:
hi brvdboss,
Thanks for mentioning the PR, the problem you mentioned we didn't consider before.

Rather than moving to the center first, we think that moving directly above the first probe point and then bringing the Z axis down would shorten the linear module movement a bit.

Wishing you and your family a very Merry Christmas.

Best Regards
Scott

brvdboss, 2020-12-27:
Hello, Iǘe made the requested change.  To be honest, I'm not exactly sure about the selection of the first (x,y) coordinates as the difference between the do_blocking_move_to_logical_xy() function and the do_blocking_move_to_xy() function.  As well as how and where the off-set for the sensor to the nozzle comes in the picture.

So currently there still is a small horizontal movement just as the calibration sequence started.  Please feel free to make additional adjustments yourself or point me in the right direction to get the 100% correct values.

closed 2023-04-04 by scotthsl