Snapmaker/Snapmaker2-Controller #37: Job Pause in CNC mode forgets Z-Position

fra-orolo, 2020-12-23:
I run a simple CNC job and wanted to pause work for fixing clamps.
After pressing Pause on touch screen the head went into up position.
When I pressed resume, the drilling bit went back to the correct X-Y coordinate to continue carving. 
But for the Z-Coordinate it went down to 0 on the the waste plate. I was lucky that the material was only thin plywood, which simply broke away instead of breaking the bit.

scotthsl, 2020-12-24:
Hi fra-orolo,
Thank you very much for your feedback!

First of all, I'd like to confirm with you if the Z-axis was out of step when you set up the work origin? If not, I would like to ask you to provide us with the log of your machine and tell us the approximate time of the problem and the name of the Gcode. Thanks! You can go as "Settings --> Export Log" in the touchscreen and send the files to me.

Our testing engineers have encountered this problem earlier, but we found that this is not a firmware bug. After analysis, we found that when the test engineers set the origin, especially when controlling the Z-axis down until the drill touches the workpiece, they tap the "Down" button too fast, which makes the Z-axis move down continuously even the drill touches the surface. Because the drill bit touches the workpiece, the Z-axis cannot actually continue to move downward, so there is an out of step operation, which causes a deviation between the real Z-axis position and the theoretical position recorded by the firmware. And this misoperation leads to an error in the set Z origin.

Suppose the real Z-axis origin of the workpiece is 20mm, but the firmware will consider the Z-axis origin value is 15mm or even lower because the out of step. When the work is paused, the toolhead will move up,  the Z-axis regains the end-stop position value at this time. When work resumes, the Z-axis will move down to the position with coordinates of 15mm and continue to work (the position recorded when setting the origin), but since the real  Z-axis origin on this workpiece should be 20mm, the drilling bit will move lower than the workpiece.

Scott
Regards

fra-orolo, 2020-12-24:
Hi Scott,

actually I had the problem you described with the 'down jog hits waste plate' before the restart. So probably the machine coordinates where off by some missed Z-Steps.  But I adjusted the workpiece start position already in the shifted coordinate system.
It could be safer for the machine, not to HOME the coordinates when going to PAUSE? (Servo motor hardware would also be great).

For the jog system on the touch controller: I think the 'keyboard buffer' or command buffer is not clever or helpful. The keys should be locked until the final position is reached, or at least the jog commands in queue should be cleared, if the next jog goes into the opposite direction. Especially with CNC milling bits you can destroy bits quite easily.
(When the milling bit hit the waste plate, I had to press the power switch on my plug connector to stop things)

Best regards
Martin

scotthsl, 2020-12-25:
Hi Martin,
Thank you for your advice.

About not homing the coordinates when going to PAUSE, it does solve the problem we talked about before.

However, when loss power and then recover the work, the machine has to return to the home position once, otherwise it can't find the reference position, and it can't go to the X, Y, Z position when loss power. But after homing the coordinates, the previous wrong Z axis ORIGIN will make the drilling bit drill more distance down.
We will improve this process after fully consider if there still are other safety risks.

About the jog system on the screen, the issue you mentioned, our product engineers have listed it as an item for improvement. We will lock the button and wait for the last motion command to be executed before the user can continue to click the button.

Wishing you and your family a very Merry Christmas.

Best regards
Scott

nonno-cicala, 2022-09-24:
I had the same problem, but I wasn't lucky and I broke the bit ☹️
Using firmware version 1.14.3
I'm not sure if my Z-axis was out of step. If that was the case it should be fixed by homing and setting the origin avoiding any contact that could cause the offset, right?

zhangquanyi, 2022-09-26:
Hi nonno-cicala,
Thank you very much for your feedback!

If the offset is not correct it should be fixed by homing and setting the origin.