Snapmaker/Snapmaker2-Controller #38: Feature Request: notification platform

baukeplugge, 2020-12-25:
Would be great to have the ability to setup notifications via email or push messages.
so you can send notifications when:
- you run out of filament
- the print is finished
- some other error occured

scotthsl, 2021-01-06:
Good idea!
We do intend to enable the machine to be monitored and controlled remotely, but have not started this work yet.

voidrin, 2021-01-06:
Thanks for the request.

I think it's nice to implement this feature, but we need to consider it carefully for personal privacy issues or some other circumstances.

baukeplugge, 2021-01-07:
Would be great! If I can help testing or discussing the options, let me know!

tigercjn, 2021-02-13:
It could all be handled externally if the controller would send an informative message out the serial port. Octoprint all ready will notify you of the print finished via pushbullet, works well for me, we just need the filament run-out to send a message out the port.