Snapmaker/Snapmaker2-Controller #42: Crash in fabscreen following run boundary

halr9000, 2020-12-27:
Steps to reproduce

1. Uploaded 3 .nc files over wifi from Luban
2. On controller, disconnected from wifi
3. Select file, press ready
4. Set x and y origins
5. Press “run boundary”

At this point the controller crashed with the error “fabscreen has stopped”. After recovering, the files I had uploaded appeared to be missing. I cycled power and found that the files were still there.

ref forum thread: https://forum.snapmaker.com/t/fabscreen-has-stopped-error-after-attempting-to-run-boundary-in-cnc-mode/13440

voidrin, 2021-01-06:
@halr9000 
We've reproduced this issue and we will fix it in the next version ASAP. Thanks for the report.

voidrin, 2021-02-03:
Hi @halr9000 , sorry for replying late.
We've release v1.11.4 firmware update (including HMI) on our [forum](https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/12) and official website. We fix the Run Boundary crash issue in this release. Please try the latest release.

ifish86, 2021-08-02:
I'm running into this problem using Fusion 360, Luban 3.15.1 and Snapmaker 2.0 A350 V1.13.1

npoirey, 2021-09-27:
same here, with a fresh reinstall of latest version. I fail on a simple 2D contour from fusion 360

connwork, 2023-09-01:
I just had this same error happen with fusion360, Luban 4.8.2, Snapmaker 2.0 A350 V1.16.7. 

Is there an actual fix to this?