Snapmaker/Snapmaker2-Controller #50: Filament Runout output to serial

tigercjn, 2021-01-05:
When the Snapmaker 2.0 detects the filament runout by the sensor going open could it send a message out the serial port so Octoprint could detect it? If it sent  "echo:busy: paused for user"  then the plugin called "OctoPrint-PauseForUserEvent" could be used to detect that user intervention is required.

GeraldPape, 2021-01-05:
Second that

scotthsl, 2021-01-06:
> When the Snapmaker 2.0 detects the filament runout by the sensor going open could it send a message out the serial port so Octoprint could detect it? If it sent "echo:busy: paused for user" then the [...]

Sorry we don't know much about the plugin "OctoPrint-PauseForUserEvent". 
Do you have a requirement to be able to resume a job that was paused due to filament runout when working online with OctoPrint?

tigercjn, 2021-01-06:
The problem at the moment is if you are running Octoprint on the Snapmaker 2.0 and you run out of filament it just keeps printing, the filament run-out detection does not flag any problem. The machine does not pause.

tigercjn, 2021-01-06:
> Got it. Just want to confirm 2 points firstly:
>
>    1. Did you disable filament runout checking by "M412 S0" in the Gcode?
>    2. If you try to print the model with touch screen? if no, please help
>    confirm if the issue also occurs in printing with touch screen. thanks.

A few people are having the same problem and its mentioned in a few areas,
here is one
https://community.octoprint.org/t/snapmaker-2-0-a350/27455
[image: image.png]

I will do some testing when it finishes the current print and confirm what
happens.

I did check the hardware switch and it does change state when the filament
it runs out, changes from Triggered to Open.

tigercjn, 2021-01-06:
There are some more comments here.
https://forum.snapmaker.com/t/sm2-0-no-filament-runout-with-octoprint/6725

tigercjn, 2021-01-07:
I tested the filament runout and it works ok if the file is sent to the screen and recovers ok. would be better if it wiped the purge off.
If running from Luban over wireless no notifications on the screen or on Luban and the head goes to the home position to load.
If running Luban connected directly it does not sense the runout and just keeps printing in mid-air. This is the same problem with running direct from Octoprint.

tigercjn, 2021-01-12:
I think we need this part of  Configuration_adv.h  uncommented.
[image]

Temennigru, 2021-01-13:
I'm copying my comment from the snapmaker forum.

It looks like it's standard practice for marlin machines with built in filament sensors to pause the print and send an action command back to the host when printing through serial

> It’s really going to be dependent on the firmware that the printer is using. If the filament runout sends an action command back to connected clients (like octoprint) that a pause has been done, then yes, OctoPrint will actually pause as long as the bundled Action Command Prompt Support plugin is not disabled.

From https://plugins.octoprint.org/plugins/actioncommands/ 1

> As of the latest Marlin bugfix-1.1.x version, the command M118 can be used to echo output to serial, meaning that M118 //action:dostuff would cause this plugin to try to handle the dostuff command when //action:dostuff was echoed back on the serial console.
> 
> A future application of the plugin would be to add actions in the RepRap machine’s firmware so that hosts could respond. For example, when Marlin is killed because something is wrong it would echo //action:poweroff and the attached OctoPrint instance could use this action command to trigger the printer’s power outlet to switch off.

From the octoprint action commands plugin:

> OctoPrint will, by default, respond to three action commands: action:disconnect (disconnects the printer), action:pause (pauses streaming the print job), and action:resume (resumes streaming the print job).

so you would want something like M118 //action:pause

tigercjn, 2021-01-20:
We just want the marlin firmware in the firmware to send out a message, using the host action commands that Marlin has would be great.  It's then up to the host, in this case, Octoprint to stop sending, alert the user and wait for the go ahead.

Schnapsmaker, 2021-01-20:
I think its nessecary to react of filament runout almost in Luban and therewith octoprint as well.

tigercjn, 2021-01-21:
Yes if you are driving the Snapmaker from a laptop running Luban this
should be able to handle filament runout, any features should be kept in
line with the Marlin standard.
Then it would work with any other complaint product like Octoprint.

chaparround, 2021-01-26:
This feature is a game changer and should be accessible by OctoPrint

pegak, 2022-02-11:
Any updates on this? :)

bf8392, 2022-08-04:
+1 I want to integrate it in Octoprint

thalesfsp, 2023-07-04:
Forget, Snapmaker is dead. Go to a real brand (Ultimaker, Prusa, etc)

dikey84, 2023-11-01:
I don't understand why this function can't just be activated quickly? Can you explain?

fachat, 2024-07-18:
Any update here?