Snapmaker/Snapmaker2-Controller #53: Feature Request: auto level calibration with heated bed and nozzle

jomjol, 2021-01-08:
**Request:** Can you please give the option not to automatically set the nozzle and bed temperature to 0°C, if you start auto bed level calibration directly from the maschine.

**Background:** The bed does bent, if you heat it up to 85°C by about 300-400µm to the outer compared to room temperature. This makes a big difference in printing adhesion, especially with object with a big footprint. 
I lost about 1kg of ABS and 2 working days, until I found the reason for the problems with adhesion, although I have tried a lot of variants with auto bed leveling.

**Work Around:** Connect to the SM2 with Luban and serial port. Then set the nozzle and the bed temperature. If you then start the auto calibration, it will hold the setting and you can do leveling unter printing conditions (heated bed and nozzle). After this procedure my large prints immedeatly where successfull.

Thanks in advance for condisering this request,
  jomjol

scotthsl, 2021-01-08:
Hi jomjol,
Thank you very much for your suggestion.

We are actually looking at this issue and we are trying to solve it from two aspects. On the one hand, we are trying to reduce the deformation of heating bed from the mechanical structure; on the other hand, we are also doing the leveling comparison test between heated and unheated.

When the test is finished, we will evaluate the test results, if the test results indicate that the heating and then leveling to obtain better printing results. We will provide an option to level after heating directly on the touch screen.

Thanks!
Scott

jomjol, 2021-01-09:
Hi Scott,
  thanks for the fast feedback. I appreciate, that you are already working on this issue. Maybe two additional remarks:
1. Reducing the deformation by structural optimization is great, but does not help the already shipped tools, like mine
2. I even would not need an additional option. If I could set the nozzle and bed temperature by hand before leveling and it just would not reset them to zero would already be sufficient.

I need to heat up the nozzle before leveling anyway to clean it, otherway I have a peace of hard filament, which will disturb the leveling anyway.

Bsherba, 2021-01-10:
We were talking about this on the forum

As we know, bed leveling depends on the temperature of the bed.

The calibration routine disables the heater, which we believe is to prevent the current flow from affecting the sensing range of the prox sensor.

Lets modify the calibration routine to wait until temperature is reached, move to position, disable bed, probe, enable bed move to postition, disable bed, probe.... etc

We were thinking of doing it ourselves but thought i'd put it out there....

scotthsl, 2021-01-12:
> Hi Scott,
> thanks for the fast feedback. I appreciate, that you are already working on this issue. Maybe two additional remarks:
>
> 1. Reducing the deformation by structural optimization is great, but [...]

In latest firmware(v1.8.0.0+), it won't abort the heating when start auto-leveling.

scotthsl, 2021-01-12:
Hi Bsherba,
Thank you for your concern!
In addition to improving the subsequent products, we are also working on improving the products that have already shipped.

We are working on improving the leveling effect right now to improve the print quality. We have experimentally verified that the heating bed brings a relatively large deformation after being heated, and its impact is already greater than the error of the sensor due to heat.
So leveling after heating is our next optimization step, and we will provide the settings on screen to choose whether to heat before leveling.
And you mentioned that adding a delay time so that after heating bed is heated to the target temperature, it will delay for a period of time before automatically starting leveling. This is a good suggestion, I will pass it on to my colleagues responsible for screen application development and suggest them to add this setting option.
Then you mentioned above that luban can't get the real time temperature after starting leveling, this is because leveling blocks the response of other gcode, and luban is getting the temperature through M105.

mdl2021, 2021-05-10:
> We are working on improving the leveling effect right now to improve the print quality. We have experimentally verified that the heating bed brings a relatively large deformation after being heated, and its impact is already greater than the error of the sensor due to heat.

Hi Scotth, I was reading the Marlin documentation and there are some settings that give you the best of both worlds. Marlin allows you to heat up the bed, then turn heaters off while you problem with the inductive sensor, then turn heaters back on while you move to the next probing position. Here is link to those settings. Maybe this can be part of the Snapmaker solution to have both heated bed without noise on the inductive sensor when probing.

https://marlinfw.org/docs/configuration/configuration.html#probe-with-heaters-off

greengranolaguy, 2021-05-10:
> Hi Scotth, I was reading the Marlin documentation and there are some settings that give you the best of both worlds. Marlin allows you to heat up the bed, then turn heaters off while you problem [...]

Hi mdl2021, this is really interesting! Do you know how we can reconfigure these settings in the Marlin configurations for the SM2? I'm new to this but it sounds like you need to recompile Marlin and then likely upload that into the firmware of the SM2. Is there a tutorial on how to do that?

peterwaespi, 2022-03-13:
I saw another 3D printer which is doing like this:

1. start g-code of model
2. printer is heating bed and extruder till the given temp of the used  filament
3. printer starts an automatic calibration before  start printing
third. 
4. print first a test line at the front end of the bed
5. starts printing the model