Snapmaker/Snapmaker2-Controller #54: Networking Issues

achampion, 2021-01-08:
Firmware: 1.10.1
Luban: 3.11.0 (3.12.3 / 3.13.0)
Laptop: MacOS Catalina (10.15)
Router: Asus AC88U

I'm having some problematic connectivity issues.

Connects to the wrong wifi... there's no option in the UI to forget a previous wifi network and because of the challenges I was having I changed to a different wifi network. However, when I restart it connects to the old wifi network.

When connected to wifi the device keeps going unreachable. It claims to be still connected to the wifi but does not respond to pings (or connection requests from Luban). Or, more confusingly, will respond (ping) to one device but not the other device (e.g. will not respond to a ping from the router but will respond to a ping from my laptop or vice versa).

E.g.:
```
[I] 08:11 me@laptop ~
[0]# ping 10.0.1.8                                                                                                                                                              08:11:19
PING 10.0.1.8 (10.0.1.8): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
```
5 mins later with no changes:
```
[I] 08:16 me@laptop ~
[0]# ping 10.0.1.8                                                                                                                                                              08:16:22
PING 10.0.1.8 (10.0.1.8): 56 data bytes
64 bytes from 10.0.1.8: icmp_seq=0 ttl=64 time=229.912 ms
64 bytes from 10.0.1.8: icmp_seq=1 ttl=64 time=46.247 ms
64 bytes from 10.0.1.8: icmp_seq=2 ttl=64 time=84.399 ms
```
Note: these are extremely long ping times.

And then some random time in the future it will stop responding to pings and luban disconnects. Sometime later it will start to respond to pings again. And the cycle repeats.

```
[I] 08:29 me@laptop ~
[0]# ping 10.0.1.8                                                                                                                                                              08:29:27
PING 10.0.1.8 (10.0.1.8): 56 data bytes
ping: sendto: Host is down
ping: sendto: Host is down
Request timeout for icmp_seq 0
ping: sendto: Host is down
Request timeout for icmp_seq 1
```

Because of this it seems to be very problematic to "discover" so I've had to downgrade to Luban 3.11 in order to be able to manually connect and not get a blank screen (https://github.com/Snapmaker/Luban/issues/640)

ywabiko, 2021-01-25:
+1 for this feature (forget Wifi).
Also, please show MAC address on Settings -> About Machine.

These two are very important when you are using MAC address filtering.

efedel, 2021-02-21:
Another vote for forget-wifi. MAC addr would be nice.

There's a common new-machine workflow that goes something like this:
  1. connect to the wide-open guest Wifi network, troubleshooting as necessary
  2. connect to the secure Wifi network once step 1 succeeds
  
 When doing this on the Snapmaker, any problems (bad password, timeouts) with the secure network cause a default connection to the guest network. If all computers are on the secure network, Luan etc cannot connect to the Snapmaker.

Another problem with the controller wifi setup is that selecting a protected network requires re-entering the password every time. Maybe that should be the long-press option when a password has already been stored.

mr-manuel, 2022-10-02:
>     2. When connected to wifi the device keeps going unreachable. It claims to be still connected to the wifi (and still prompts to "disconnect" if previously connected to luban) but does not respond to pings (or connection requests from Luban). Or, more confusingly, will respond (ping) to one device but not the other device (e.g. will not respond to a ping from the router but will respond to a ping from my laptop or vice versa).

I have exaclty the same issue since the update from Luban 3 to 4 and update from the Snapmaker 2.0 A250 firmware to the latest version. The Snapmaker has 4 meters distance to the WiFi Access Point and there is no wall or other objects in between. I'm not able to print directly from the PC anymore. I have to copy the files to the USB and then plug the USB in the Snapmaker. Could we please investigate furter to solve this for all?

szimonek, 2023-03-26:
I have my WiFi router standing 2 meters above Snapmaper 2.0, and machine can't even find my wireless networks, but finds networks from my neighbours. This ticket wasn't even answered within over two years. This is unacceptable. 

Maybe we all should fill a support issue to stop being ignored by snapmaker support?