Snapmaker/Snapmaker2-Controller #55: Feature Request & Safety Measures CNC machining "Go to Work Origin" button

blockmodule, 2021-01-10:
Feature Request & Safety Measures

Hello!
I was setting up to prepare for CNC machining.
I was setting up to prepare for CNC machining, and while I was getting used to it because I had done a lot of CNC machining, I ran into a problem.

Because of my large hands and the small size of the touch panel operation, I had to press the wrong button (
I pressed the Go to Work Origin button right next to the Set Work Origin button) and broke the end mill.

If it was only broken, it would be fine, but it would lead to damage to the CNC rotation axis and module as well as the X and Y axes.
In addition, the possibility of injury from the broken end mill cannot be denied, so please adopt this as a safety measure.

Countermeasures
If the origin of the machining center position and Z position is not set on the touch panel (Set Work Origin), the "Go to Work Origin" button cannot be pressed.

Trouble can be prevented by allowing the buttons to be pressed in the order in which the settings are completed.
1. Start CNC machining settings. 2.

2. Move XYZ position ("Set Work Origin" button: Enabled "Go to Work Origin" button: Disabled Boundary confirmation button: Disabled) 3.

3. Determine the machining center position and end mill calibration position. Press the "Set Work Origin" button ("Set Origin" button: enabled, "Go to Work Origin" button: enabled).

Translated with www.DeepL.com/Translator (free version)

sdj544, 2021-01-10:
Sorry but this is a terrible idea.
If you do any cnc with multiple tool paths and bits you need to be able to use go to work origin quite often and you don't want to have the origin changing or have to re-enter it. 
What is needed to be added is an emergency stop to the screen.
There is a physical emergency stop button coming out along with the rotary module.
In the meantime any time you're working with cnc you need to stay in reach of the power button. 

-S

blockmodule, 2021-01-10:
Hello, sdj544!

I hadn't thought about the task of replacing the bit.
When I thought about that task, I understood that changing it would be problematic.

I guess I'll withdraw my change request!

Ps. I wonder if there is a way to prevent pressing the wrong button....

closed 2021-01-10 by blockmodule