Snapmaker/Snapmaker2-Controller #58: File size issue (gcode > around 250Mb)

MlvPerry, 2021-01-13:
Hello, I have an issue with gcode files (for 3d printing) larger than approximately 250Mb.
They do not finish transfering to the Snapmaker 2.0 A350, or they crash the display software before starting the print.
I tried with different slicers and different files. For some prints is not possible to further reduce the file size via resolution decrease or print settings optimization...
I see other users have the same or very similar problems, for example [here](https://forum.snapmaker.com/t/file-transfer-issues/8618) and [here](https://forum.snapmaker.com/t/snapmaker-2-0-cura-profile/7537).

scotthsl, 2021-01-21:
Hi MlvPerry,
Thanks for your feedback!

My colleagues  sad this is known Luban issue, they will fix it.

closed 2021-01-25 by scotthsl

ITmaze, 2021-01-26:
If this is the case it's customary to link to the issue in the other tree
that you can find what's going on in relation to it being fixed.

MlvPerry, 2021-01-26:
> If this is the case it's customary to link to the issue in the other tree that you can find what's going on in relation to it being fixed.

@ITmaze Yes, that sounds like good practice... Sorry, but I have very little experience with GitHub and related protocols, should I just link the issue in the Luban repository? (as a new issue) Or the correct way is a different one? Or, yet, it's better if a collaborator does something about it?
I couldn't find much info online about it, but maybe I'm just using the wrong search terms... (?? [https://github.com/marketplace/actions/add-an-issue-link](url) ??)

This problem is currently preventing me from using the Snapmaker 2.0 as a printer alltogether.
Thank you for your help!

ITmaze, 2021-01-26:
My comment was directed at the person who closed the issue.

The way to create a link is just to paste the URL into a comment.

Given that the issue was closed by scotthsl, it would be their
responsibility to create the link, not yours, MlvPerry.