Snapmaker/Snapmaker2-Controller #61: Possibility of manual Z leveling

Gspohu, 2021-01-17:
Hello,

First of all, thank you for your great work!

An important improvement would be to have the possibility to do the leveling manually and especially with more points than auto-leveling.
Indeed, a lot of local leveling problems could be avoided with this technique, for example, I level my other machines (CNC and 3D printer) with a comparator and the precision is unequalled! It is within everyone's reach a comparator costs around 20€/$. But I haven't found any way to do this with the Snapmaker 2.0. However, my comparator tells me that it would be necessary.

This feature would be as important for 3D printing as it is for CNC mode, indeed this week I wanted to engrave PCBs in Engraving and accurate leveling is very important in this case, otherwise the near tracks won't be correct.

Have a nice day,

scotthsl, 2021-01-21:
Hi Gspohu,
Thank you very much for your advice!

First of all I am  interested in your mention of comparator, I would appreciate if you could provide me with a photo or a link to it.

Then you mentioned manual Z leveling, which is supported by snapmaker2.0. You can turn off auto leveling in the setting of 3DP, and then go to Calibration, at which time the machine will performs manual leveling.

Best Regards
Scott

Gspohu, 2021-01-28:
Thank you for your answer.

Here a link of a comparator : https://www.micronfrance.com/comparateur-digital.html

Thank you for your advice, but it seem work only on 3D print and not with CNC, as I said I would like to engrave PCB, so the leveling is really important.

For now, I try to do some leveling through Gcode, but I have some trouble with G29 command because this command always want to go under the bed.

Gspohu, 2021-01-29:
Hello,

Here is my Gcode for CNC leveling :
```
G28;Home
M420 S1;Set leveling on
G92 X-19 Y347 Z200
G01 X131 Y105 Z200 F2000
G29 L131 R167 F105 B162 V4
;Xmax 326 Ymax 357 R=L+36 B=F+57
M420 V;Display mesh
M500;Save mesh
M501;Load mesh
M420 V;Display mesh
M421 I0 J0 Z0
M421 I0 J1 Z0
M421 I0 J2 Z0
M421 I1 J0 Z0
M421 I1 J1 Z0
M421 I1 J2 Z0
M421 I2 J0 Z0
M421 I2 J1 Z0
M421 I2 J2 Z0
M420 V;Display mesh
M500;Save mesh
M501;Load mesh
M420 V;Display mesh
;G4 S10;Pause 10 seconds
G28 ;Home
G92 X0 Y347 Z334
M420 S1;Set leveling on
G01 X131 Y105 F2000;Point 0
G01 Z74 F500;Height active comparator
G92 X0 Y0 Z0
; Point 1
G42 I0J0 F2000;Go to mesh point I0 J0
G4 S5; Pause 5 second
G0 Z20 F3000
G1 Z0.1 F100
; Point 2
G42 I1J0 F2000;Go to mesh point I1 J0
G1 Z0.1 F100
G4 S5; Pause 5 second
G0 Z20 F3000
; Point 3
G42 I2J0 F2000
G1 Z0.1 F100
G4 S5; Pause 5 second
G0 Z20 F3000
; Point 4
G42 I2J1 F2000
G1 Z0.1 F100
G4 S5; Pause 5 second
G0 Z20 F3000
; Point 5
G42 I1J1 F2000
G1 Z0.1 F100
G4 S5; Pause 5 second
G0 Z20 F3000
; Point 6
G42 I0J1 F2000
G1 Z0.1 F100
G4 S5; Pause 5 second
G0 Z20 F3000
; Point 7
G42 I0J2 F2000
G1 Z0.1 F100
G4 S5; Pause 5 second
G0 Z20 F3000
; Point 8
G42 I1J2 F2000
G1 Z0.1 F100
G4 S5; Pause 5 second
G0 Z20 F3000
; Point 9
G42 I2J2 F2000
G1 Z0.1 F100
G4 S5; Pause 5 second
G28;Home

```

And it works quite well, except at the time of the G29 declaration (*G29 L131 R167 F105 B162 V4*). I have to remove the bed at the time of the declaration of the G29 (I am too attached to my Snapmaker to take the slightest risk), because it makes the head go down to Z0, without taking into account the coordinate system updated by G92, I think G29 only use machine coordinate. But the G42 (G42 Ix Jx F2000) respect the G92, so either I have a bad understanding of the Gcode (it's possible I'm still a novice with the Gcode), or there's a bug. I've done a lot of research on this subject and I haven't found anything very conclusive, except for this bug fix from Marlin (https://github.com/MarlinFirmware/Marlin/issues/20587), but I'm not really sure since it's a bit old.

**Why am I making this line: *G29 L131 R167 F105 B162 V4* ?**
As I said I want to engrave PCBs, so I need to do very localised leveling (just on the surface of my PCB), so this line allows me to declare a new set of 9 measurement points in this area.
Which I save with M500 and activate with M501 and then the G42 commands go to the right coordinates. So I write down the values in a notebook and integrate them into the leveling in the Gcode of my CNC engraving with M421 I1 J1 Z0.33, for example for point I1 J1. If my method seems bad to you to reach my goal, do not hesitate to let me know.

**Why am I presenting this Gcode here?**
Both to look for a solution to my little problem with G29, but also to give some food for thought to the implementation of such a feature in the CNC part of the Snapmaker firmware.

**How to reproduce the problem ?**
Test with this minimal Gcode :
```
G28
M420 S1
G92 X-19 Y347 Z200
G01 X0 Y0 Z200 F2000
G29 L131 R167 F105 B162 V4
```

**:warning: Be careful if the problem is the same on your side, you must remove the bed before the tests to avoid any damage to the machine.**
**Good to know : ** G29 uses the last executed position as the starting point, this is why there is the line *G01 X0 Y0 Z200 F2000*.

Version of my firmware: **V1.10.1_20200822**

Thank you for reading me and thank you for your great work.

gomezag, 2021-04-04:
Thanks for the code @Gspohu !
I wanna try this, but I wanna be sure to understand the procedure.. Can I just save this code to a .gcode file and load it to luban? 
Before starting, I have to take off the bed for the first G29 command, and after it goes to the first levelling position reinstall it and continue from there right? 
Thanks again!