Snapmaker/Snapmaker2-Controller #63: Feature request: Snapmaker Linear Module Inversion

ShadowsNight1000, 2021-01-21:
Could we get an option in the Snapmaker to invert the X axis module? It would really help with moving the control module to the other side of the printer and keep the cables away from the Enclosure door.

scotthsl, 2021-01-21:
> Could we get an option in the Snapmaker to invert the X axis module? It would really help with moving the control module to the other side of the printer and keep the cables away from the Enclosure door.

I'm a little confused as to which scenario you are talking about to move tool head to the other side? Couldn't you move the tool head to the other side now?

ShadowsNight1000, 2021-01-21:
It would allow flipping the Linear Modules so the cables came out the other side. I couldn't really see a point for the Z and Y axis, but for the X axis it would mean the control board that is on the right hand side, could be flipped and used on the left hand side. It would give a little more freedom with the power supply placement and exhaust duct outlet on the enclosure. and stop the cables from opening the side door on the Snapmaker enclosure.

ITmaze, 2021-01-21:
It's not about changing the tool head side, but bolting the controller to
the other Z-axis rail.

The X-axis would need to be flipped over, putting the stepper (and cable)
near the other Z-axis, which would make that it's working backwards.

The biggest concern is that the coordinate system for X would be reversed
and homing would go to the wrong corner.
--
finger painting on glass is an inexact art - apologies for any errors in
this scra^Hibble

()/)/)() ..ASCII for Onno..

ShadowsNight1000, 2021-01-22:
ITmaze has worded it better than me, I'm still very noob at anything to do with a 3d printer, as I haven't even printed yet.

scotthsl, 2021-01-25:
Yes, I understood what you mean. But how do you place the touch screen if you mount the controller on the other Z axis?
The bracket for touch screen can only be assembled on the right side (the right side when you are facing the machine).

ITmaze, 2021-01-26:
We're not talking about flipping over the entire enclosure, just making
more space next to the door.

The touch screen stays where it is.

ShadowsNight1000, 2021-01-26:
There's plenty of cable for the screen to reach to the other side of the Snapmaker, the plan is to use a couple cable clips on the enclosure to take the strain off the USB-C port from the coiled cable.

jenkins007, 2021-01-29:
Why can't you unscrew the controller and screw it on the other axis side and reroute the cables to the other side?

ITmaze, 2021-02-01:
That's what this feature request is about.

We can make the physical conversion, but the x-axis will be reversed.
Updating the software in the controller will make it possible to achieve
this change.

jonflacke, 2021-04-21:
I, too, would like this. I do not have an enclosure and likely never will. However, my printer sits in a left corner on my desk. With the controller being on the right Z-axis, the wires all hang out into my work area and run the risk of getting snagged and damaging the system. To have them all on the left would mean they're facing a wall - much lower risk of getting snagged as I work on things. (picture for added clarity)
[image]
I've already printed up a mount for my screen to place the screen on top of the left Z-axis so I can get around that problem easily enough, though it would've been nice to have the screw holes for the screen mount on both sides giving the option of a left or right Z-axis controller placement. The only thing that is not possible to overcome at the moment is having the system set up for an inverted X-axis.
An added benefit of having the controller on the left Z-axis side will be that the cable to the 3D printer head and the 3D filament will no longer cross each other.

ldp7, 2021-04-22:
@jonflacke  Nice try on modifying the touchscreen holder. As for the inverted X-axis, our firmware developer, @xhycode , said we can add some new G-code command to achieve it. We have received your request and will add this feature before May. Considering the testings and bug-fixing, we think it can be released in May. We will update it in the changelog.

Cheers
Edwin

Voodoofly, 2022-04-03:
The same, have a own housing

I want to have the controller on the left side 

Have a F350 Machine and will assembled the controller to the left side, mechanical works all fine
The X-Axis Linear Modul must flip, when not flip the cables are to short

But when flip, then is left - right from x-Axis reversed

so you have more profit from the installation options with the software alone

I hope for a Machine Option, when assemled to say Controller right side, or left side
Please  make us a “X-AXIS Reverse Option” in the F350 Machine

I see now, more people wish the option, im not alone... 

is it so difficult to install it in the firmware?

Controller right Side (Standard)
Controller left side (Reverse Option)

snips56, 2022-06-13:
Is this change now implemented in firmware? How do you set up for a flipped X axis module?

I am also in the same position - just the configuration of my workshop and a home-built enclosure that needs the controller moved to the other upright. Alternatively - is there a longer cable/ cable extension available for the X axis for A350 machine?

If this is not available in FW, is there a g-code instruction we  can add in to files to get reversed right-left movement on the X-axis?

haddood, 2023-07-24:
Hi,
Can anyone confirm if this feature has been implemented. I am in the same situation like all above, I'd like to put controller on the left side for my setup (flipped enclosure)

rdif, 2024-02-17:
Hello. Please let me know if I am addressing the wrong issue. I simply want to place the filament spool on the same side as the filament hole is placed on the heating head, which is the right side. In the current configuration, the heating head cable is rubbing against the filament. I have identified two problems with this: the length of the screen cable and the length of the X-axis cable. I am fine with the first one, as we can easily install the screen holder on the other side. However, the second problem seems to persist even after extending the cable, causing issues with the placement of this cable. The most convenient solution would be to mirror the X-axis in the firmware. I have found some [configurations](https://github.com/Snapmaker/Snapmaker2-Controller/blob/cc2e81a8030042c6564ea0f9060f15b66717818c/config/default/Configuration.h#L957) to do so, can anyone confirm if changing the INVERT_X_DIR will work in my case?