Snapmaker/Snapmaker2-Controller #64: Unable to start a new job after stopping a previous one

uniederer, 2021-01-21:
With Snapmaker 2, Firmware V1.10.1_20200822, i run regularly into the situation, where a print job is started, but temperatures are set to 0 after a few seconds.

Steps to reproduce:

1. Start a 3D printing job
2. During heat up phase (and I believe also during print) hit "Stop"
3. As soon as the UI allows for starting a new print, start any other print
4. Observe the temperatures:
   a.  The new set points are according to the new part
   b. After a couple of seconds, the set points drop to 0, the controller remains in "heatup" mode.

The only way to persuade SM2 to start another print I've seen so far is to powercycle the whole machine.

Does anybody else have this issue as well?

daniel-starke, 2021-01-23:
Same issue on my side. Also with G-code generated by PrusaSlicer.

scotthsl, 2021-01-25:
Thanks for your feedback!

Based on your description, that should be a known issue. When you don't start heating the nozzle from room temperature, it may trigger a heating failure. That's because the V1.10.1 firmware's thermal protection is too sensitive, and next release will fix that. The temporary solution for now is to wait until the temperature of the nozzle drops to room temperature before starting to heat it up again.

uniederer, 2021-02-07:
@daniel-starke Oops... didin't notice that one. Thanks for linking. It looks indeed related.
@scotthsl After upgrading to the latest firmware just released lately, it appears indeed, that the issue is gone. I'll do some extra testing again and then report back.

scotthsl, 2021-02-09:
Hi Uniederer,
If this issue reappears, please provide log of the console output about the nozzle heating.

To record the logs:
You could connect the machine to luban's console, or other console tool, and then keep the console open to reproduce the issue.

Thanks a lot!
Scott

closed 2021-08-13 by xhycode