Snapmaker/Snapmaker2-Controller #65: Feature Request: Turn on enclosure light by default upon startup.

ywabiko, 2021-01-25:
Please add an optional setting that turns on enclosure light by default during startup from power on.

scotthsl, 2021-01-25:
May I ask why you want the light bar to be turned on during power on?

brent113, 2021-01-25:
I agree with this being an option.

It's dark in the enclosure and the first thing I always do is turn the light on.

In the meantime I just compiled a custom module firmware where the enclosure lights are on by default when the conclosure gets power.

ywabiko, 2021-01-25:
Like brent113, I always turn the light on manually every time Snapmaker boots up, simply because it's too dark inside due to its orange acrylic, and I want to always see the inside to double check safety.

jenkins007, 2021-01-25:
I too have to manually turn the LED lights on when powering the snapmaker 2.0 on. This option will save extra clicks to turn it on which is more user friendly and a better UI. Making it an option does not seem like it would take much time or energy to adjust the firmware. I'll be happy to test it as I was a test lead for Clear Channel Communications.

scotthsl, 2021-01-26:
Hey guys, thank you so much for telling us your feedback on your experience. The product manager has just confirmed with me that this feature will be added.

Another feature will be added as well: changing the tool head while keeping the enclosure light on. By the way, thanks to @brent113  for adding a M2001 command to the firmware, touch screen will use this command to temporarily turn off the power supply of the 8-Pin ports so that the tool head can be replaced safely.

tHeSmUrF, 2021-03-18:
@scotthsl  Adding to this I would like to see an option where the lights come on automatically when you open the enclosure door(s).

xhycode, 2021-03-18:
We confirmed with the product manager that this feature will also be added, thank you very much for your suggestion!

olishell, 2021-04-17:
> @scotthsl Adding to this I would like to see an option where the lights come on automatically when you open the enclosure door(s).

> Hey guys, thank you so much for telling us your feedback on your experience. The product manager has just confirmed with me that this feature will be added.
>
> Another feature will be added as well: [...]

Have any of these features been implemented? Lights on by default upon power up and the ability to change tool heads with lights on would be fantastic.

iamrecursion, 2021-05-05:
I'm also interested in when these might be coming!

voidrin, 2021-05-28:
> Lights on by default upon power up and the ability to change tool heads with lights on would be fantastic.

We will implement lights on automatically when power on first,  touchscreen will have an option to control it. We are working on this feature now, please stay tuned.

Bpositivedotdk, 2021-07-12:
Yes great idea :-)

roymar2, 2021-12-05:
I am very interested in this feature also. I have working on a wifi system to turn them independent of whether snapmaker is turned on or not. Doing this would mean disconnecting the light strip from the controler.

xhycode, 2022-01-11:
This issue has been supported in [V1.13.2](https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/17) and later, please use the latest firmware.

closed 2022-01-11 by xhycode