Snapmaker/Snapmaker2-Controller #66: Using Luban to start printing automatically using "Send to Device via Wi-Fi"

jenkins007, 2021-01-25:
Add the capability to start a print using Luban, but having the status of the print job showing on the Snapmaker touch screen. My computer room is far from the printer and I don't stay in there unless I am actively using the computer. My Snapmaker is in my game room which I can see from the kitchen and the living room, so the status of the print job is very important to me to be on the touch screen. This may need to be addressed in both the software forum and the firmware forum. 

3D printing is the most important aspect of this request, although it could also be used for Laser and CNC, with the expectation that the user (me) knows the paths are clear on the workspace and safe. I always have the Snapmaker 2.0 A350 ready to print before sending any file to it with the enclosure ready as well.

tHeSmUrF, 2021-03-18:
+1 to this request.  My printer is in another building and it would be great to be able to send from my PC and the job starts without needing me to walk the 50 metres to my shed.  I'm the same as jenkins007 - I always leave my printer in a ready to go state.

Bpositivedotdk, 2021-07-12:
+1 to this request

MIIB1, 2022-07-17:
+1 to this request