Snapmaker/Snapmaker2-Controller #69: Reset of calibration after power cycle.

nfmaas, 2021-01-28:
Hi, Edwin mailed me that i should post this here.
I have the problem that when i shut the printer down and start it up again the next day my calibration is gone. All is set to the default 9.00 . I allready got a replacement controller and print head. But it keeps resetting to default.
When i am in the console function over the com port and do a M501 it recalls all the calibration and after a M500 it will run fine. But it doesn't load it automatically. Any idea what causes this? My gues it is in the touchscreen as that is the only thing left with some ai brain in it.

Regards, 
Nand

nfmaas, 2021-02-02:
[image]
[image]

scotthsl, 2021-02-09:
Hi nfmaas,
Sorry for late reply!

I don't know why I can't view the attachment you uploaded, is this problem you are talking about occurring with the latest  firmware?

The problem of not being able to record settings should not be due to power off, it is possible that there is a problem when saving.

Can you provide the log of following operations.
1. connect the machine to luban's console, or another console tool, and open the console
2. Execute the auto leveling, the console will show the leveling log at this time
3. wait for the leveling to finish, adjust the z offset on the touch screen, and then click save.
4. At this time, first enter M501 in the console, and then enter M503.
5. Send all the logs on the console back to us

Thanks a lot!
Scott

nfmaas, 2021-02-09:
> Hi nfmaas,
> Sorry for late reply!
>
> I don't know why I can't view the attachment you uploaded, is this problem you are talking about occurring with the latest firmware?
>
> The problem of not being able [...]

Hereby Scott,

This is what you requested.

Strangely enough after i performed the M1999 it didn't fall back to default. I have now just turned the power off,  An on again after a few minutes but now it recalls everything. So i don't know why this is happening. Just last week and the weeks before that it changed to 9.00 every time i turned it off.

[autocal_M501_M503.txt](https://github.com/Snapmaker/Snapmaker2-Controller/files/5950282/autocal_M501_M503.txt)

scotthsl, 2021-02-09:
Hi nfmaas,
Thank you for your prompt reply.

From your log, everything looks fine: After leveling is done, controller will write the new leveling data to the its flash. when you enter M501, controller will reload the data just written from flash, and then from the output of M503, the data is correct.

Then you enter M1999, which will reboot the controller, and then controller will also reload the leveling data from flash, but you said the data was correct.

However, after shutting down the controller for a few minutes, when you turn it on again, the data fall back to default and here I feel very strange.

Could you please help to try the following operations to see if the data will be reset?
1. Re-level and get the new leveling data
2. turn off the power
3. wait 5s, turn on the power
4. connect console, input M503, please send us the log output from M503

Perform the above operations, mainly want to see if there is a difference between turning off the power for 5 seconds, and turning off the power for a few minutes. Tanks a lot!

Best Regards
Scott

nfmaas, 2021-02-09:
> Hi nfmaas,
> Thank you for your prompt reply.
>
> From your log, everything looks fine: After leveling is done, controller will write the new leveling data to the its flash. when you enter M501, [...]

Hereby Scott, i did upgrade the firmware just now

[NEW M503.txt](https://github.com/Snapmaker/Snapmaker2-Controller/files/5950841/NEW.M503.txt)