Snapmaker/Snapmaker2-Controller #70: Controller freezes

jenkins007, 2021-01-29:
Using SM 2.0 A350, I sent a file to the SM and started the print job on the SM. I then noticed that the LED lights were off and looked at the touch screen to see if I could turn them on. I could not (that is one issue).

I went to the computer and connected to the SM. I turned the lights on. I want to ALWAYS see the status of the print job on the touch screen and clicked disconnect on the computer. I went to the SM and clicked disconnect. It would not disconnect. I noticed the bed and the nozzle temps were not ready to print yet. I then went back to the SM and it stated that if I disconnected the print job would be cancelled. ??? The job started on the SM. That should not happen. I attempted to click on confirm (to disconnect on the SM). After a few minutes the touch screen showed this error message... FabScreen keeps stopping. I clicked below the error message that stated to close the app and it restarted.

scotthsl, 2021-02-02:
Thanks for your feedback!

May I ask what is your current firmware version number?
And then can you provide the gcode file you were using, as well as the screen log?

jenkins007, 2021-02-02:
A350 with V1.10.1 firmware. Using 3 different G-code models.  Steps to recreate:
1. Send gcode from Luban to A350 via WiFi
2. Tap start print on A350 (I noticed the temp on the nozzle was 0 degrees, I sent the job to the printer the night before)
3. Go back to Luban and click connect
4. Turn on/off LED (this may not be necessary, did not try without doing this step)
5. Click cancel print on A350
6. Click confirm (I pressed 5 times)
7. After 5+ minutes click confirm many times
8. After a few more minutes, click outside the "Confirm" area
9. Confirm goes away, click cancel print job
10. FabScreen error appears 
[FW_0.log](https://github.com/Snapmaker/Snapmaker2-Controller/files/5911236/FW_0.log)
[SC_0.log](https://github.com/Snapmaker/Snapmaker2-Controller/files/5911237/SC_0.log)
[SC_1.log](https://github.com/Snapmaker/Snapmaker2-Controller/files/5911238/SC_1.log)
[Filament Filter Oiler.gcode.txt](https://github.com/Snapmaker/Snapmaker2-Controller/files/5911172/Filament.Filter.Oiler.gcode.txt)

scotthsl, 2021-02-03:
Thank you very much for the information!
We will try to reproduce the issue according to the steps you provided.

scotthsl, 2021-02-09:
Hi jenkins007,
Sorry for late reply!

This issue was caused by a collaboration problem between luban and touch screen, our engineers have analyzed the root cause.

Thanks again for your feedback, we will fix this issue as soon as possible.

Best Regards
Scott