Snapmaker/Snapmaker2-Controller #71: Release improvement: Can you also use the releases or tags for new versions of the firmware

brvdboss, 2021-02-03:
A new release of the firmware is published and was announced on the forum: https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/12  (maybe through other ways like facebook too but I don't have a facebook account).

But it would be nice if the github releases also lists these.  Just like how it's done for Luban. Then in the announcements a full list of improvements & changes can easily be generated or referenced through the issue numbers.
Now there are some nice features not highlighted (like the power domain management that could be used to allow switching tool heads while the machine is powered on etc.)

I know this spans the controller repo; the modules and the unreleased code for the touch screen.  But now it's impossible to know up till which commit is included in the public release.  (For example, is the power domain change I mention actually included or not?  I would have to test & reverse engineer).

At least adding a tag marking the release would be a good idea.

brvdboss, 2021-02-03:
ok, looks like none of the commits from January were included in the newly released firmware?

scotthsl, 2021-02-04:
Hi brvdboss,
Thank you very much for your concern!

First of all, I'm sorry this has caused trouble for you.

This release is not based on code from this repository due to historical legacies. Actually the firmware for controller in this release was ready before we opening the source, just waiting for the update of screen and then released together.

Another reason is that screen firmware does not yet support the emergency stop and rotation module, so the controller firmware from this repository cannot be released together with the screen firmware yet.

But we promise that from the next release, the controller firmware will be made with this repository (meanwhile the old private repository will be abandoned and we don't want to maintain two repositories at the same time), the tag and release note will be made like luban.

Best Regards
Scott

scotthsl, 2021-02-04:
> ok, looks like none of the commits from January were included in the newly released firmware?

For this release, there are 3 changes for controller:
1. CNC bug: if delta of Z axis from current position to maximal position is less than 30mm, spindle won't be turn off while stop/finish current work.  commit: 80e5357c07b97779ee64cb7b09455f363fe694ee
2. optimize heating protection: e939c23ecb997ed945b5e35ca4641ba63ffe89fb
3. reduce sensitivity of detection for screen offline: because the refactoring, there is no separate commit for it, but location of the function is in snapmaker.cpp, line 201-217:
``` 
    if(READ(SCREEN_DET_PIN)) {
      xTaskNotifyStateClear(task_param->heartbeat);

      if (systemservice.GetCurrentStatus() == SYSTAT_WORK && count) {
        // if we lost screen in working for 100ms, stop current work
        systemservice.StopTrigger(TRIGGER_SOURCE_SC_LOST);
        LOG_E("stop cur work because screen lost!\n");
      }

      if (++count)
        count = 0;

      vTaskDelay(pdMS_TO_TICKS(100));
      continue;
    }
    else
      count = 0;
```

brvdboss, 2021-02-04:
Hello Scott,

Thank you for your very quick and elaborate response! I appreciate that!
And don't worry, it didn't cause any  real problems for me. I'll roll my own for now, but I didn't expect to be doing a downgrade from the version I was running already ;)

Looking forward to the future releases!
I'll close this issue with this comment

brvdboss, 2021-03-27:
I suggest we reopen this issue again until it's actually done.
still regularly a question about this on the forum

scotthsl, 2021-03-30:
> I suggest we reopen this issue again until it's actually done.
> still regularly a question about this on the forum

 👌

brvdboss, 2021-12-13:
Reacting to this again after a new firmware release (and some months of silence):

Is it possible to add tags to this repository to at least indicate which code is included in the new firmware release.  Now it's always guessing what is included in the firmware and what not.

Just adding a tag firmware Snapmaker2_V1.14.1 would create a lot more transparancy and is in my opinion a small effort with no implications on the software development process

Thanks

xhycode, 2021-12-13:
Hi brvdboss
The  [V4.4.11 tag](https://github.com/Snapmaker/Snapmaker2-Controller/releases/tag/v4.4.11) has been updated

brvdboss, 2021-12-13:
@xhycode Thanks. 
An additional question:
Where is it linked to the actual firmware version nr?

In this case:
Snapmaker2_V1.14.1

Is it documented somewhere what version of controller software is included in that particular firmware version (which also includes the UI, modules firmware etc)?