Snapmaker/Snapmaker2-Controller #73: :feelsgood: After power Off/On always hook to WiFi that was used during first setup and no Forget Wi-Fi option

tamer1an, 2021-02-04:
Expected: 

1) After power Off/On machine connected to the same WiFI network that was connected previous 
2) Or at least Forget Network option could be an faster solution

I had stock firmware before updating to the latest Snapmaker2_V1.11.4 114
https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/12?u=tamer1an

Before: 
[image]

Machine always connected to the same WiFi that was active before shutdown
After wake up it was connected to the same WiFI network again

After:
[image]

After Power On machine often connecting to the network that was used on initial unboxing stage or stays not connected to WiFI at all and I have to type a complex password every time :finnadie: and there is no forget WiFi network option

KoolFrank87, 2021-02-08:
I too am having this exact problem.  The A350 is forgetting the WiFi password after power cycling. Very annoying. :-(

samvanh, 2021-02-08:
Same issue on latest firmware. Wifi connection does not persist across restarts.

samvanh, 2021-02-09:
Don't think this is an HMI issue but rather a software bug.

kahuwi, 2021-02-15:
Same issue on latest firmware. Wifi connection not remembered.
This is a QA oversight that needs to be fixed ASAP

dusekdominik, 2021-02-17:
Same issue.

thejerry1, 2021-02-18:
Same issue.

samvanh, 2021-02-18:
To the developers: the issue doesn't seem to be in password retention but rather a bug in the initial connection. 

Expected use: 
Switch on the snapmaker, wait a couple of minutes, connect automatically to last wifi network

Actual:
No wifi connection established. 

2 possible workarounds:
1. Settings -> wifi -> select wifi network and type in passwords again
2. Settings -> wifi -> toggle wifi switch off and on.  Wait and couple of secs and it will connect to last network without prompting for password.

dwienie, 2021-02-22:
@samvanh  the second workaround you mention  will that solve the issue forever or  you need to do that every time the wifi connection is not established?

samvanh, 2021-02-22:
> @samvanh the second workaround you mention will that solve the issue forever or you need to do that every time the wifi connection is not established?

Every time... But at least you don't need to re-enter the password

kahuwi, 2021-02-22:
This sequence needs to happen automatically on system power up.   When the snapmaker is not close at hand, having to go to the unit to manually connect to wifi is NOT the way network based system work in the real world.  On my setup, the power is turned on remotely and the snapmaker is in another building away from my computer.

dwienie, 2021-02-22:
> This sequence needs to happen automatically on system power up. When the snapmaker is not close at hand, having to go to the unit to manually connect to wifi is NOT the way network based system work [...]

Totally agree

tlaurion, 2021-03-13:
Same issue.
Workarounds are to be done at each reboot:

> To the developers: the issue doesn't seem to be in password retention but rather a bug in the initial connection.
>
> Expected use:
> Switch on the snapmaker, wait a couple of minutes, connect [...]

ncurley999, 2021-03-20:
I'm having the same issue unfortunately.

tamer1an, 2021-03-26:
Thanks for the report guys, I found a corresponding issue in the software repo. 

https://github.com/Snapmaker/Luban/issues/788

Please leave ur comment also there to escalate this further

slipperysam1984, 2021-03-29:
Same issue on A250 :-(

genemyers, 2021-04-22:
was this ever addressed?

NiteCrwlr, 2021-06-10:
I'm having the same problem. A350, no wifi connection after machine reboot. Latest version Snapmaker2_V1.12.0.

tabledevil, 2021-07-02:
Same Issue on my A350. Is there a way to downgrade the firmware again?

tamer1an, 2021-07-24:
Looks like it was fixed with the latest update, check it out guys. Re open in case if it still not working for you.

closed 2021-07-24 by tamer1an