Snapmaker/Snapmaker2-Controller #74: M303 PID Autotune doesn't work

ypt0i, 2021-02-06:
every time attempting M303 the controller get stuck in a loop that only solved by powering off the machine.
tried both via WiFi and via Serial connection..

always getting same error of thermal runaway error :
"Error: Heating failed, system stopped! Heater_ID: 0 - thermal run away of hotend!"

(true - actual temperature doesn't go up)

*also tried with the recent firmware V1.11.4

scotthsl, 2021-02-09:
First of all, it should be explained that the printer's PID does not support M303 adjustment for now. Printer's heating process is controlled by itself, not by controller. We will later improve the M303 command so that it can tune the printer's PID.

Also for the heating failure problem you mentioned, can you provide a complete console log of the heating process?

Thanks a lot!
Scott

ypt0i, 2021-02-15:
certainly, here is the log:

> echo:Unknown command: "�]0M1005"
snapmaker-luban 3.13.1
Connected to COM5
G53
ok
G28
others < leveling OFF
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
others < leveling ON
others < Apply Z offset: 0.05
X:-15.50 Y:253.00 Z:231.68 E:0.00 Count X:0 Y:104000 Z:94000
ok
> M303
M303
others < PID Autotune start
others < new E target temp: 0
others < new B target temp: 0
others < new E target temp: 0
 T:18.10 /0.00 B:19.37 /0.00 @:127 B@:0
 T:18.20 /0.00 B:19.49 /0.00 @:127 B@:0
 T:17.70 /0.00 B:19.80 /0.00 @:127 B@:0
 T:18.80 /0.00 B:20.00 /0.00 @:127 B@:0
 T:17.80 /0.00 B:19.30 /0.00 @:127 B@:0
 T:18.40 /0.00 B:19.69 /0.00 @:127 B@:0
 T:18.00 /0.00 B:19.26 /0.00 @:127 B@:0
 T:18.40 /0.00 B:19.77 /0.00 @:127 B@:0
 T:18.00 /0.00 B:19.45 /0.00 @:127 B@:0
 T:17.80 /0.00 B:19.41 /0.00 @:127 B@:0
 T:17.80 /0.00 B:19.41 /0.00 @:127 B@:0
 T:18.40 /0.00 B:19.22 /0.00 @:127 B@:0
Error:Heating failed, system stopped! Heater_ID: 0
others < thermal run away of hotend! temp: 18.40 / 0
others < new E target temp: 0
others < new B target temp: 0
others < cannot stop in current status[1]
 T:17.80 /0.00 B:19.61 /0.00 @:127 B@:0
Error:Heating failed, system stopped! Heater_ID: 0
 T:18.20 /0.00 B:19.84 /0.00 @:127 B@:0
Error:Heating failed, system stopped! Heater_ID: 0
 T:17.40 /0.00 B:20.04 /0.00 @:127 B@:0
Error:Heating failed, system stopped! Heater_ID: 0
 T:18.30 /0.00 B:20.00 /0.00 @:127 B@:0
Error:Heating failed, system stopped! Heater_ID: 0
 T:18.20 /0.00 B:19.49 /0.00 @:127 B@:0
Error:Heating failed, system stopped! Heater_ID: 0
 T:18.00 /0.00 B:19.80 /0.00 @:127 B@:0
Error:Heating failed, system stopped! Heater_ID: 0
 T:18.10 /0.00 B:19.26 /0.00 @:127 B@:0
Error:Heating failed, system stopped! Heater_ID: 0
 T:18.20 /0.00 B:19.61 /0.00 @:127 B@:0
Error:Heating failed, system stopped! Heater_ID: 0`

and the loop goes on until I cut off the power.

as for the PID, so the values stored in the controller (M301 command) doesn't affect the hotend as well?

scotthsl, 2021-02-20:
Sorry for late reply!

for your question, yes, hotend does not use the PID stored in controller, but the one stored on the toolhead. 

Also from your log, the PID tuning process is problematic, we will optimize it later and implement new M303 so that it can tune hotend's PIDs.

brent113, 2021-03-16:
It appears M301 is correctly implemented to send PID values to the toolhead. However, the CAN function ID 16, which has different names in the controller (MODULE_FUNC_3DP_PID) and module (FUNC_REPORT_TEMP_PID), is not implemented in the controller.

The typical way of retrieving PID values would be with M503, but M503 reads the PID parameters in the controller only.

Perhaps an easy way to correct this mismatch would be upon system startup the toolhead reports the PID parameters to the controller, then the controller can set its local copy of the PID parameters so they are guaranteed to match.

xhycode, 2021-03-17:
Thank you for your suggestion, it is a very good suggestion. Since the heating of the module is controlled by ourselves, we did not optimize the controller to set the PID, we will optimize this subsequently

M503: Synchronize the PID stored in the module

M303:Function added

M301: optimize the setting of PID operation (currently can not be used directly)

gkress, 2021-04-13:
xhycode and all,
Are you saying that the M303 function is now added?  I just downloaded and installed the V12.0 Firmware and the M303 is still not working.  I am getting the following errors that eventually timed out to:

Error:Heating failed, system stopped! Heater_ID: 0
others < PID Autotune failed! timeout
others < new E target temp: 0
others < new B target temp: 0
others < new E target temp: 0

I don't necessarily understand brent113's comments about M301 and M503.  Is the still no way to do a PID check?  Is is really necessary to do a full Snapmaker 2.0 A350 3D printer calibration?  My prints are NOT as good as I feel they should be and I am trying to check everything.  Any suggestions?

TheIvoryDisaster, 2022-09-09:
Is there any update on this? What are the recommended steps to tune customized hot ends made with different materials? I am seeing temperature fluctuations in my build I would like to be able to control more accurately.