Snapmaker/Snapmaker2-Controller #76: Feature Request: Support G2/G3 arcs

constant-flow, 2021-02-08:
Hi, when using the CNC post-processor of FreeCAD you also provide i realised, that arcs `G2` and `G3` are not supported. I read that your Fusion360 post-processor breaks the arcs into linear moves `G1`. What is the temporary approach for me for now as well. 

But, why isn't G2-3 supported, as the firmware is based on Marlin, if i understand correctly.

Is there a technical/hardware limitation?
Else it should be more or less one line to enable it: https://marlinfw.org/docs/configuration/configuration.html#g2/g3-arc-support

Thx in advance 😄 

_tag me as_ [`Firmware: General Marlin`]

tigercjn, 2021-02-16:
They did change something a few years ago.  Minimum segments for G2/G3. Better for small arcs. (#13466)
so it must have been though of.

zauguin, 2021-02-16:
@tigercjn That change does not come from Snapmaker but from upstream Marlin, so it doesn't necessarily imply that Snapmaker thought about it.

@constant-flow On my Snapmaker 2.0 G2 and G3 work correctly, but I never tried it through Luban. Have you tried sending the commands directly?

tigercjn, 2021-02-17:
Great, they should work then, someone else had layer shifting being caused
by they think G2 and G3. I will give it some testing.

constant-flow, 2021-02-17:
Sending codes via console work but when included in a cnc file they behave weird very slow/pause (maybe the `F` wasn't included but the firmware was searching for it)

When sending the codes manually, the machine sounded weird for slower arc lines (almost like step skipping).
I never investigated deeper into this, as i thought devs might had a reason to suggest linear movements over arcs (see fusion360)

For the meantime I adjusted the post-processor in a fork of this branch. It's quite a codechange so not sure if i should open a PR - but it converts all curved movements to tiny segments and also accepts non-marlin drill commands (use care fully as spindle speed is too high for deeper+bigger holes ♨️ ). And creates a picture of the job.

scotthsl, 2021-02-20:
Hi all,
Thanks for feedback！

We didn't test G2/G3 before, but  controller firmware will be optimized for this later.

infoteknix, 2021-03-18:
Hi, can I add my observations?: 
I recently did a simple CNC job, milling a small internal rectangular profile with radiused corners. Corner arc radius generated is only 0.5mm (G2).  Milling feedrate 600mm/min. Controller virtually stops when processing the corners - then takes off again for the linear sections.  Checking the G2_G3.cpp code, is it possible that the call to thermalmanager.manage_heater() on every iteration of the segments (line 163) is causing slow calculation time? Could it move to the idle handling below so it's not called on every iteration?
I might, however, be out of my depth here!

Sample G-Code for one depth pass:
; 2mm 2 flute Endmill
M3 P100
G0 X-104.5 Y-22.5
G0 Z1.0
G1 F300.0 Z-1.0
G1 F600.0 Y-20.0
G2 X-104.0 Y-19.5 I0.5 J0.0
G1 X-47.0
G2 X-46.5 Y-20.0 I0.0 J-0.5
G1 Y-22.5
G2 X-47.0 Y-23.0 I-0.5 J0.0
G1 X-104.0
G2 X-104.5 Y-22.5 I0.0 J0.5

Mark

dpeart, 2021-03-22:
I've noticed this exact slow behavior when trying to 3d print using G2/G3 arcs.

I use the OctoPrint plugin to create gcode with arcs, but it isn't usable because of this issue.

ldp7, 2021-04-22:
Request received and we have added it to the request pool.

rstocks, 2021-08-21:
Just to add a bit of context to this: I was chatting with the author of ArcWelder (processes gcode to convert points into arcs), and his quick review of the Snapmaker firmware indicated that it's an older version of Marlin, which has some issues with the arc commands. Merging in the current version should resolve the "slow behaviour" when using G2/G3 commands.

If you need an example, I've popped some samples here: https://www.amazon.com.au/clouddrive/share/5xn1PvIHdet8E5VtRsMonCFFwOHPEZ3xphCGzJMRguF. There's the stl file for a model of a few curves, the gcode as generated by Luban, and the gcode as processed by ArcWelder.

I've got a video side-by-side comparison of printing the original and the "welded" gcode here: https://youtu.be/_qbOlGby4Pc (yes, it's over an hour long - the first 60 seconds should show you everything you need to see!)

tigercjn, 2021-08-21:
Well done on showing the benefits of AcrWelder and also of using the latest
Marlin updates, with data like this action should be taken and
released ASAP by the team.

Great input.
Chris

JostTim, 2021-12-22:
Just a little info on the slow G2/G3 moves, in case the merge with current marlin version is delayed because it can bring some other troubles on the stack.
A quick fix in the meantime could be this : 

>  Recommended to change files ``G2_G3.cpp#L219`` and ``G2_G3.cpp#L239`` and replace ``seg_length`` by 0.0

Source : MarlinFirmware/Marlin#17348

dapendragon, 2023-03-21:
@ldp7 Any updates on support for this?

dapendragon, 2024-08-19:
@scotthsl @ldp7 any news on this? Eagerly awaited! 😉