Snapmaker/Snapmaker2-Controller #78: Please make HMI code available

mrworf, 2021-02-14:
As a avid android developer (having worked on BSP for mobile phones and other devices) I'd love to be able to tweak and play with the android version running on the touch screen. In other words, if possible, please share repo for building android for the HMI 😄 

Thanks

scotthsl, 2021-02-17:
Thank you for your concern!

I have forwarded this issue to my Android development colleagues.

rbcapp, 2021-12-27:
Bumping this request. There is a request for 11 X 11 bed leveling made in the facebook group. After looking though the controller codebase, it looks like the only modification needed is to change the HMI to provide the option and send the request to the controller. Opening up the HMI codebase would allow for enhancement without using your development resources.

DragonNZ, 2022-07-28:
Bumping this request too. As another android developer, I agree that we could help enhance the HMI code without loading up your development resources.

mofe23, 2023-01-01:
Another bump - Dear Snapmakers, please consider open sourcing to empower your community to own their devices end to end!

Robdirt, 2023-02-23:
Bump, doesn't make sense to me for this to be the only thing that snapmaker is not willing to let the community to contribute to.

Bsherba, 2023-10-21:
I would also like to see the HMI available. I want to contribute by adding an option to the settings for a parking position for filament change while in the enclosure as it is very inconvenient to do so from the natural home.