Snapmaker/Snapmaker2-Controller #79: Request: Please make the software STOP button on the controller an interrupt.

nissguy, 2021-02-15:
I keep having issues where I'm going to have a crash, or did and I hit the stop button to no avail, only to have to use the main power button.

It seems that the stop on the controller just sends the gcode stop command but must wait until the current operation is finished.

If I'm pressing stop, I don't want to stop when the machine is done running across the work platform. I'd like it stopped immediately.

Thanks!

scotthsl, 2021-02-17:
Thank you for your feedback, I'm sorry this has caused you trouble.

We would like to know that the stop button you mentioned cannot abort  current work. Does it mean when heater is heating up, or when machine is working in printing?

Thanks a lot!
Scott

mr-manuel, 2021-03-16:
I have the same issue, when the maschine is working in printing. Maybe a solution can be to add another button with stop immediately?

tedjianlichen, 2021-03-17:
Sorry, this trouble bothers you.
we'll solve it asap

cocojoe, 2021-07-05:
@scotthsl I think the stop should mean STOP under any use case. It should be an interruptive instant STOP like a `kill -9`

nissguy, 2021-07-05:
@scotthsl 

Sorry I didn't see this earlier and respond. The issue I've been having is when the machine is moving. If I screw up the work coordinates, or try to move it using the movement controls but press the move button too many times. It would be nice if when the stop button was pressed it stopped doing whatever it was doing and just stopped.

Thanks!

The3rdIcon, 2022-12-16:
This would be a great alternative to having to restart the machine after you hit the button.