Snapmaker/Snapmaker2-Controller #85: Future Request: Operation upon stopping at filament detection

blockmodule, 2021-03-10:
Future Request: Operation upon stopping at filament detection

Since there were no objections to my suggestion in the Snapmaker forum, I would like to make a feature request for the following

Forum
https://forum.snapmaker.com/t/filament-that-is-running-low/15464?u=blockmodule

Background
When molding for a long time, there is a possibility that the user is away from the machine or away from the Snapmaker machine, and may not be aware of the filament replacement.
In this case, it would be better if the nozzle temperature can be lowered and standby is possible when there is no operation for a certain period of time, because leaving the machine for a long time may result in a decrease in heater life or even a fire.

Objectives
1. fire prevention by maintaining heat nozzle temperature
2. prevention of nozzle burnout
3. prevent filament from falling out of the nozzle naturally
4. countermeasure for heater life

Contents
When the filament is lost and the filament sensor reacts.
With the current latest firmware (Snapmaker2_V1.11.4), the machine will return to the home position after stopping and wait for the operation instruction on the touch panel.

Problems
The nozzle temperature remains elevated while waiting for instructions.

Improvement request
When the waiting time for instructions on the touch panel exceeds 10 minutes, return the nozzle temperature to room temperature.
When an operation instruction is entered on the touch panel, the nozzle temperature is raised and after reaching the temperature, the filament is replaced and ejected, and then molding is continued.
The temperature effect is only on the nozzle temperature.
The temperature of the heat bed is kept at the same level to avoid the possibility of the molding object peeling off.

xhycode, 2021-03-10:
Thank you very much for your suggestion, this issue has been added to our requirement pool, and this feature will be added subsequently

blockmodule, 2021-03-10:
I know it is a lot of work to develop this product, but I would appreciate your cooperation.

dotnetfx, 2022-04-04:
In addition to it being a high temperature safety hazard, when the filament runs out and the nozzle remains hot, if left unattended, the nozzle will also burn up the filament remaining in the nozzle causing it to carbonize and then jam.  When this happens the only solution is a complete replacement.  For very long prints it's quite possible for the filament to run out in the middle of the night while unattended and thus causing the nozzle failure because of the filament burn up situation.  Effectively rendering the runout option useless.  I've come across this scenario.