Snapmaker/Snapmaker2-Controller #88: Future Request: G-code line and code display showing processing progress

blockmodule, 2021-03-14:
Future Request: G-code line and code display showing processing progress

Since there were no objections to my proposal in the Snapmaker forum, I would like to make the following feature request.

Forum
https://forum.snapmaker.com/t/display-of-g-code-lines-and-codes-to-show-machining-progress/16777?u=blockmodule

Background
In general functions of general-purpose CNC or machining center machines, the execution code during machining is displayed on the control panel in real time, and the exact number of G-code lines and codes at the stop position are displayed for emergency stop, sensor stop, and arbitrary stop. However, Snapmaker does not display the execution code information at the time of stop, and recovery is not possible.

Purpose
To display the information visually on the touch panel.
2. to display the processing G-code when some kind of stop occurs.
The display should show the line number and G-Code where the G-Code was stopped.
If possible, display the code at the time of execution plus one line before and after, for a total of three lines on the touch panel during execution.
5. display for 3 types of 3D printer/laser/CNC.

Details of the proposal
Display three lines including one line before and after the current G code on the touch panel during 3D printer/laser/CNC processing.
Display the line number of the G-code and the code of the stop position for all the stop states that occur during execution. If possible, we would also like to display the stop position when power is lost and power is recovered.

Possible solutions
In a situation where it is not possible to continue to start the machine because the position where the machine stopped for some reason is currently unclear, being able to grasp the stop position will enable the machine to re-edit the G-code and restart.
This will prevent the loss of molding time and workpiece assets.

xhycode, 2021-03-16:
Thank you for your suggestion, we have understood your need, now you can use M2000 S0 to query the current line number after manually pausing. Check the "Last st line:" information.

    However, the query after reboot is not supported yet, and the line number queried after stopping may have deviation. The firmware will optimize the line number saving function later

blockmodule, 2021-03-17:
Hi,xhycode 
It will be a relief for many users to see the running G-code on the touch screen and know what code has been stopped.

We apologize for the many improvements that continue to be made, and thank you for your patience in this matter.