Snapmaker/Snapmaker2-Controller #90: Automatic power supply fan speed

mr-manuel, 2021-03-16:
Hi,

the fan of the power suffly is spinning the whole time at 100%, even the Snapmaker 2.0 is idle and the power supply is producing no heat. Is it possible to turn off the fan if the Snapmaker is idle? Greater would even be if it's possible to control the power supply fan by the internal temperature.
E.g.:
* No heat = fan off
* Some heat = fan at 50%
* * Hot = fan at 100 %

xhycode, 2021-03-17:
I am very sorry to give you a bad user experience.
The fan of the power supply is hardware controlled. So your suggestion cannot be realized.
Sorry again for the inconvenience!

mr-manuel, 2021-03-17:
Ok, thank you. Maybe it would be an idea to optimize hardware for future builds :)

xhycode, 2021-03-17:
This is a very good idea, we will optimize the power in the next version, thank you for your support!

scotthsl, 2021-07-10:
New power module will enter mass production. so close this issue.

closed 2021-07-10 by scotthsl