Snapmaker/Snapmaker2-Controller #98: Feature request: add z offset option that's independent of calibration

Temennigru, 2021-04-10:
Whenever you need to raise or lower the z offset, you have to go through calibration all over again. This should not be necessary. Given the fiddly nature of z offset, it takes a few experiments to get it right, and being able to set it independently is necessary.

ldp7, 2021-04-13:
@Temennigru 

When the machine is printing, you can left slide the touchscreen and adjust the Z offset in real-time.

See the pictures below:
[image]

[image]

Cheers 
Edwin

Temennigru, 2021-04-13:
That is print-specific and not a general setting tho.
For instance, I can’t change the general z offset and print via print-server.

stewartwlewis, 2021-07-02:
You can also do this from a terminal in Luban or for example,  Octoprint.  use G1029 Dx.xx where a negative x.xx moves the head closer to the bed. Positive moves it up.  So G1029 D-0.05 is the same as a Z offset of -0.05.  The only difference is you dont have a number you adjust you are just moving the levelling mesh up or down.  This works with the Z adjustment detailed above - both are effectively applied.

Temennigru, 2021-07-02:
I haven't had any luck with G1029, but M206 Z0.05 (coordinates offset) has worked for lowering the bed for me

stewartwlewis, 2021-07-02:
I find G1029 D works ok, check the bed level matrix with M420V, do a G1029 D adjustment and check M420V.    I will have changed and this matrix is what is used when printing.  I don't like M206 as the only way to get it reset is a level from the touch screen or a machine reset with an M1999. I do all levelling with G1029 and a macro that has the sensor Z offset in it so this doesn't work for me but great you have got something that does work for you.

Temennigru, 2021-07-02:
You can just M206 Z0 (or whatever it was before) to reset it