Snapmaker/Snapmaker2-Controller #99: Feature request: Add e-step calibration to touchscreen

Temennigru, 2021-04-10:
e-step calibration is an important step to getting consistent prints. It should not be required that you use manual console commands to do it, and therefore should be supported in the touchscreen

ldp7, 2021-04-13:
Thank you for your request.

As we all know that the e-step calibration is important to the printing quality and consistency. But both Snapmaker Luban and the touchscreen do not have the option for adjusting this item. We will discuss it with our mechanical engineer and make sure if we can develop this feature in the near future.

It will need the cooperation between the stepper motor, firmware and the touchscreen UI. We will report the progress in the changelog. Please stay tuned.

Best regards
Edwin

tHeSmUrF, 2021-05-03:
+1 to this request. People (including myself) are reporting huge improvements to their prints after calibration.  Performing this calibration should be in the getting started manual as a must do and as part of the setup guide on the machine itself.

JH1616, 2021-05-11:
+1 please add the e-step calibration to the touchscreen.

nissguy, 2021-07-05:
+1 here!

scyto, 2021-09-17:
does this request mean just extruder estep or XYZ esteps too?

I found doing XYZ esteps can help alot when dimensional accuracy is required, unfortunately setting XYZ esteps breaks calibration catastrophically :-(

Learoy1, 2021-09-25:
Please let us change esteps on the machine, or at least adjust flow % during print.

Learoy1, 2022-06-23:
Still no further progress on this? Please let us control flow it's a basic but important highly useful function