Snapmaker/Snapmaker2-Controller #100: Start heating nozzle before bed gets to target temperature

jenkins007, 2021-04-11:
Using the A350, when starting each 3d print job, the nozzle does not attempt to heat until the bed reaches the desired temperature. This is inefficient. I realize it takes a long time for the bed to reach the target temp (usually 60 degrees for my prints), but the nozzle could start heating up when the bed is within a set temp range either a percentage (80% or so), or within 10 degrees or so.  This would cut off a couple of minutes of print time at least.

ldp7, 2021-04-13:
@jenkins00

Which slicer did you use? If you generated G-code file with Snapmaker Luban, the nozzle and the heated bed will heat up simultaneously. Here are the codes for your reference:
> M82 ;absolute extrusion mode
;Start GCode begin
M104 S205
M140 S50
G28 ;home
G90 ;absolute positioning
G1 X-10 Y-10 F3000
G1 Z0 F1800
M109 S205;Wait for Hotend Temperature
M190 S50;Wait for Bed Temperature
G92 E0
G1 E20 F200
G92 E0

The command "M104" set the extruder temperature and heat up immediately.

You can add the following code at the beginning of the G-code file manually.
>M104 Snnn (Snnn: Target temperature)
M140 Snnn

Cheers 
Edwin

scotthsl, 2021-07-10:
Close this issue because Gcode from luban has start heating nozzle firstly.

closed 2021-07-10 by scotthsl