Snapmaker/Snapmaker2-Controller #102: Start printing on Controller

jenkins007, 2021-04-12:
Prior to the new firmware update, I could send a file to the printer and the printer controller had a button to start printing. Now it shows "Got It" and it takes 4 "clicks" to start the print job. Please bring back the button to immediately start the print job on the controller after receiving a file.

ldp7, 2021-04-13:
Some users have reported the workflow change in Snapmaker Firmware V1.12.0.
[New Firmware for A350: Why changed dialogue? 2](https://forum.snapmaker.com/t/new-firmware-for-a350-why-changed-dialog/17551)

For Firmware 1.12.0, this is a transition phase to the Rotary Module. The workflow of transferring the file via Wi-Fi will be improved in the next version of Snapmaker Firmware.

The touchscreen will not ask you to start the work but inform you the file has been transferred successfully after you send the file to the machine.

After you tap “OK” on the touchscreen, you will navigate to the HOME page. You need to select the files in the File section and start the work manually.

Our UI team is working on an improved workflow to offer a better user experience.
For users who do not have the Rotary Module, you can continue to use Firmware 1.11.4.

Sorry for the inconvenience.

Edwin

ldp7, 2021-04-13:
Some users have reported the workflow change in Snapmaker Firmware V1.12.0.
[New Firmware for A350: Why changed dialogue? ](https://forum.snapmaker.com/t/new-firmware-for-a350-why-changed-dialog/17551)

For Firmware 1.12.0, this is a transition phase to the Rotary Module. The workflow of transferring the file via Wi-Fi will be improved in the next version of Snapmaker Firmware.

The touchscreen will not ask you to start the work but inform you the file has been transferred successfully after you send the file to the machine.

After you tap “OK” on the touchscreen, you will navigate to the HOME page. You need to select the files in the File section and start the work manually.

Our UI team is working on an improved workflow to offer a better user experience.

For users who do not have the Rotary Module, you can continue to use Firmware 1.11.4.

Sorry for the inconvenience.

Edwin

jenkins007, 2021-04-13:
It still goes back to user experience. 4 clicks versus 1. You could easily have both the "Got It" button and the "Start Print" button on the WiFi received screen.

ldp7, 2021-04-14:
> It still goes back to the user experience. 4 clicks versus 1. You could easily have both the "Got It" button and the "Start Print" button on the WiFi received screen.

@jenkins007 
We know that this workflow may be difficult and a little lengthy to users. Our product manager is designing a new workflow and that will be more user-friendly.

Sorry for the inconvenience.
Edwin

artemician, 2021-04-14:
@ldp7 Edwin I would like to add something. I've noticed on the forums that some people prefer to keep the confirmation button requirement that only took one step as opposed to four. The four steps issue aside, maybe make it a user choice option in settings to choose whether they want the confirmation button enabled or not, for those that would like to keep the confirmation button.

ldp7, 2021-04-14:
Nice suggestion and I will forward it to our product manager. 

I will keep this issue posted under this thread.

Cheers
Edwin