Snapmaker/Snapmaker2-Controller #109: Enable bed leveling post a G28 home command?

jorgeszabo, 2021-05-05:
Relatively new to this world so please excuse me if this has already been reported and addressed. 

Marlin’s documentation says that G28 (home) turns off bed leveling unless RESTORE_LEVELING_AFTER_G28 is used. Looking at the source code here, this configuration seems to be disabled. 

When looking at Gcode exports from Luban I don’t see a M420 S1 command following the G28 home command at the top of the file. Does this mean the bed leveling matrix is not loaded? Or Where is the command that tells the controller to load the bed level mesh?

jorgeszabo, 2021-05-05:
After looking into this more I realize the bed leveling does come back on after completion of a G28 command. Still unsure on where this command is being executed but I'll close this issue since it is not really a bug report or issue at this time. Feel free to delete as well.

brent113, 2021-05-05:
https://github.com/Snapmaker/Snapmaker2-Controller/blob/c09a114cfef9892fe7d2b4aa5b56531ffc828a8d/Marlin/src/gcode/calibrate/G28.cpp#L462

xhycode, 2021-08-13:
Thanks for brent113's reply, This process is directly controlled by G28 code, I will close this issues.

closed 2021-08-13 by xhycode