Snapmaker/Snapmaker2-Controller #110: Hot-end not shutting off after job! Thermal runaway issue!

DougJoseph, 2021-05-06:
At least twice since I updated the firmware to version 1.11.4_20210201 I have found the hot-end still hot, long after a print job was completed. Has anyone else had this issue?

ldp7, 2021-05-06:
Dear Doug,

Thank you for contacting us.

With the firmware 1.11.4_20210201, we have received any similar cases. 

Please update the firmware to V1.12.1, which is the latest one.
https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/14

And, you can check your G-codes. There should be some commands to stop heating.
>;End GCode begin
M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)

Edwin

DougJoseph, 2021-05-06:
My end G-Code:

;End GCode begin
M104 S0
M140 S0
G90
G92 E0
G1 E-1 F300
G1 Z330 E-1 F3000
G1 X0 F3000
G1 Y350 F3000
M84
;End GCode end

DougJoseph, 2021-05-06:
You wrote, "With the firmware 1.11.4_20210201, we have received any similar cases."

Did you intend to write, "we **not** have received" instead of "we have received"??

DougJoseph, 2021-05-06:
> Please update the firmware to V1.12.1, which is the latest one

I have now updated the firmware to V1.12.1. Will watch closely and report back.

ldp7, 2021-05-06:
Sorry for the typo. We have not received similar cases.

Your Gcode have the commands "M104 S0" and "M140 S0", which disable the heater after the print is completed.

Please keep me updated about the result with firmware V1.12.1.

Cheers
Edwin

DougJoseph, 2021-05-06:
Thanks! Will do!

DougJoseph, 2021-05-15:
OK, I'm back, with more details, and with still having issue.

After updating the firmware to the latest, I ran another print job, and then waited afterwards. The toolhead (3D print module) is staying warm to the touch, even after the nozzle has cooled down. It stays warm for as long as I leave the machine on.

I'm very concerned that something is wrong, and likely a fire hazard.

Is there anything else in that toolhead that is designed to generate heat, besides the heater for the nozzle?

I'm concerned that perhaps the first time was thermal runaway of the nozzle heater, but that the incident melted wiring insulation, etc, such that now I have a "new" issue unrelated to the nozzle heater.

xhycode, 2021-06-08:
Hi Joseph,
After upgrading to 1.12.1 firmware, the hotend hot issue is gone. Is that right?

You said the printing head keeps warm. If you let the machine on for 10mins, will it stay warm? Which part of the printing module is warm?

One more question, when the nozzle cools down, does the fan of the printing module still run? Thanks

DougJoseph, 2021-06-08:
> After upgrading to 1.12.1 firmware, the hotend hot issue is gone. Is that right?

Hello. No, the issue is not gone. As I stated in my previous reply:

```
OK, I'm back, with more details, and with **still having issue**.

After updating the firmware to the latest, I ran another print job, and then waited afterwards. **The toolhead (3D print module) is staying warm to the touch, even after the nozzle has cooled down. It stays warm for as long as I leave the machine on**.

I'm very concerned that something is wrong, and likely a fire hazard.

Is there anything else in that toolhead that is designed to generate heat, besides the heater for the nozzle?

I'm concerned that perhaps the first time was thermal runaway of the nozzle heater, but that the incident melted wiring insulation, etc, such that now I have a "new" issue unrelated to the nozzle heater. 
```
I don't yet have any further details beyond the above.

Tracyxia-tech, 2021-06-08:
Hi Joseph,
I will send you an email to follow up on the issue that you met. Please check in your inbox. Thanks

DougJoseph, 2021-06-08:
> Hi Joseph,
> I will send you an email to follow up on the issue that you met. Please check in your inbox. Thanks

Thank you. I will look for it.

Tracyxia-tech, 2021-06-08:
My pleasure :)

DougJoseph, 2021-06-08:
In the email you mentioned, "please export the log and send it to me."

How do I export the log?

Snapmaker-support, 2021-06-09:
Here is a video about how to export log. https://drive.google.com/file/d/189CIUW4D4o_Qm2mY2j63N3rKzn7n3zZp/view
The log goes to the USB stick. Thanks

DougJoseph, 2021-06-09:
> Here is a video about how to export log. https://drive.google.com/file/d/189CIUW4D4o_Qm2mY2j63N3rKzn7n3zZp/view
> The log goes to the USB stick. Thanks

Thank you.

DougJoseph, 2021-06-13:
> Here is a video about how to export log. https://drive.google.com/file/d/189CIUW4D4o_Qm2mY2j63N3rKzn7n3zZp/view
> The log goes to the USB stick. Thanks

Hey, I'm back with more info, but first I have a question about the log files I exported. Do the logs show only up to the last reboot? Or do they carry over to a time before the machine was last shut down?

DougJoseph, 2021-06-13:
I ran a print job. A couple of hours after the job was 100% completed, the screen was reporting the nozzle was holding steady between 36 C to 37 C degrees, and the printhead felt quite warm to the touch. The nozzle seemed to be about as warm. Both the printhead and nozzle were "touchable" but definitely way too warm for two hours after completion and ample time.

I pressed the "complete" button. I waited about 5-10 more minutes, and the nozzle was still holding between 36 C to 37 C degrees. The printhead was still warm to the touch. I tried to export the log, but my USB drive was not formatted properly. At that time, I shut off the power out of concern.

Later, after getting my USB drive properly formatted, I turned the machine back on, and exported the log. (If the logs date back to before a previous shut down, then I have valid logs. If the logs only date back to the the last shut down, I suppose I don't.)
  
The print job had been sliced and outputted from Cura 4.9.1, with the following printer profile "end g-code" settings:

`;End GCode begin
M104 S0
M140 S0
G90
G92 E0
G1 E-1 F300
G1 Z330 E-1 F3000
G1 X0 F3000
G1 Y350 F3000
M84
;End GCode end`  

Here's a screen shot of the Cura Printer Profile:
https://drive.google.com/file/d/1H0J0yxYOpeRB1JhXgSCsJSXfXN_WbA3v/view?usp=sharing

Here's a zip file of the log files:
https://drive.google.com/file/d/12ImaLPkv7usUY67A_2Ijn24zKuhQsXBm/view?usp=sharing

Any and all help appreciated!

DougJoseph, 2021-06-13:
Also, here are links to images that were requested via email. The first image is after 100% completion but before pressing the finish button. The second one is some time after pressing the finish button, about 5-10 minutes later.

https://drive.google.com/file/d/16LUJ5SPgt6ok15rURgH7zaOjQxZ4eW-t/view?usp=sharing

https://drive.google.com/file/d/1hpi6IgIt7jJU8hYIHI6aE03xwqjVYndq/view?usp=sharing

mdl2021, 2021-06-13:
Hi Doug,

One possible explanation is that you don't turn off your stepper motors 

with a M84 ;steppers off in the End code. (I always have M84 in my end code 
and many others do ) The extruder has a stepper motor in the head to feed 
the filament. When a stepper motor is stationary is on with its coils fully 
energised. So lets say the stepper is a 12V 1A stepper this would mean that 
it will generate 12W of heat and that has to be radiated so where so hence 
given the all-metal construction possibly that heat is being picked up by 
the Nozel thermocouple.

Try adding M84 into your end script and see if the temperature of the 

nozel drops after a print. If it does not then there may be some issue with 
the nozel heater/software.

Best regards
Lyndon

DougJoseph, 2021-06-13:
> Try adding M84 into your end script and see if the temperature of the nozzle drops after a print.

Thank you, Lyndon. However, as you can see from my end-code that I posted above, I already have M84 in my end code. Here again is an exact copy of my end G-Code:

`;End GCode begin
M104 S0
M140 S0
G90
G92 E0
G1 E-1 F300
G1 Z330 E-1 F3000
G1 X0 F3000
G1 Y350 F3000
M84
;End GCode end`

Anyhow, thanks for being so helpful.

DougJoseph, 2021-06-14:
The thought occurred to me today: what if the M84 is being ignored and the stepper motors are still engaged, i.e. still left on?

If so, it might explain the additional heat in the printhead, because of the extruder stepper motor being located there.

Also, if so, it means we would need to solve why the M84 in the "end G-code" is being ignored.

DougJoseph, 2021-06-14:
I'm currently in the middle of another long print job with the SM 2 A350. Whenever this print job ends, I will pay careful attention to whether the stepper motors are remaining engaged (in defiance of the M84 command in my end G-code) or whether they are indeed being turned off as expected. More soon.

DougJoseph, 2021-06-15:
Well, @mdl2021  (Lyndon), I have to say, you get a prize for helping me hone my grasp of the issue. I do still need help solving it though.

As I said, I have M84 in my end G-code (coming from the Cura slicer). However, the M84 command is being ignored by the printer after the job completes! Furthermore, when I use Luban to connect via wifi and send the M84 command in the terminal, it gets ignored then too. Anyone have ideas on that???

theenginerdinglab, 2021-06-15:
Hi @DougJoseph 

I'm pleased I drew attention to the M84.

I replied from my email to your original comment and my email cropped off the text with your end codes, hence why I did not see them originally. I was looking at your Cura screenshots of the end code settings which does have the M84 missing.

I will run a test today on my A350 and watch the head temperature with the extruder stepper motor on and then after an hour, I will do an M84 to see if I see any changes in the temperature. I'll update later with the results.

DougJoseph, 2021-06-15:
@theenginerdinglab: Thank you so much!

closed 2021-08-13 by xhycode