Snapmaker/Snapmaker2-Controller #113: Touch Screen  or G1029 bed levelling does not probe the points used when printing

stewartwlewis, 2021-05-17:
When printing the software adjusts the Z height in accordance with a leveling mesh that can be shown with M420V.  That mesh is calculated from saved Z offset values that correspond to logical points on the bed.  This calculation is done on the logical coordinates before any workspace offset is taken into account.  When you print with a workspace the Z offset is calculated for the point on the bed where printing occurs so you never need to worry about workspace when doing a leveling.  

Standard levelling using G29 or G42 turns off workspace but when leveling is initiated using the touch screen or g1029 the X&Y coordinates used for the Z offset measurement are those for the current workspace.  The point probed is then saved in the logical leveling mesh which means the whole mesh is off by the value of the workspace.  on an A350 the workspace used is -19 -10 as detailed in M206. 

This can be easily seen by placing some small blocks (4-5mm) on the G1029 W i1 j2 and G1029 W i3 j2 positions and do a bed level.  Then get the head to travel from X0 Y160  to X320 Y160.  The head moves across the bed it moves up and down in accordance with the levelling mesh.   The high points should be in accordance with the blocks on X91 Y170 and X227 Y170.  The low point at X159 Y170.  This is not what happens though.  The high points are actually at the logical (G29) points of X72 Y160 and X208 Y160 and the low point is at X140 Y160.
Please be careful as the nozzle will crash into the bed if you start at to low a height.  Suggest Z5.

A corrected mesh can be achieved by turning off workspace before probing but the bed coordinates have ben set in such a way that the logical bed starts from x4 y8 and ends at x276 y312 as the logical bed coordinates have been set to accommodate workspace.  

To fix the logical bed size and probing limits need to be set correctly and touch screen and G1029 leveling need to be done outside any workspace.  When sorting this it would also be useful if the probe offset values in configuration.h could be taken into account, they are currently ignored.  

Sheet showing all coordinates here.
https://docs.google.com/spreadsheets/d/13kjel1GpdRVb6PDADTlBHXG7reKYzK6_nyXHUxjk0Bk/edit?usp=sharing

ldp7, 2021-05-24:
we are reading this issue now.

stewartwlewis, 2021-05-30:
I received a response from Snapmaker Edwin which did not answer this issue so I had another attempt at explaining the problem.
https://docs.google.com/document/d/1a4EoKOw0SB-QJYbTsILFWKFY6ge0TjqiNskMM0S_SsY/edit?usp=sharing

ldp7, 2021-06-01:
Hi @stewartwlewis , I asked our software developer @parachvte to check this issue, and he is good at coding and English. I am sorry that we cannot offer the proper answer to you so far.

stewartwlewis, 2021-06-01:
Thanks, thats great.  I am happy that somebody is looking into it.  Let me know if want any further information or clarification from me.

ldp7, 2021-06-02:
Hi @stewartwlewis , here is the reply from @parachvte and @scotthsl .
https://www.notion.so/Response-from-Scott-Huang-c9cec593425344d4bb7814db5141e96a

Cheers

stewartwlewis, 2021-06-03:
Thanks @scotthuang and @parachvte for looking into this. I am not a coder but I will go thought this and see if I can understand your explanation.
I understand that you are using RAW coordinates and you think everything is fine but in practice it is not.  I don’t do gode that is why I created the practical experiment which I still think shows there is a problem that you have not explained.. 
Rather than do the experiment with coordinates I did it with lines drawn on paper taped to the bed so that I could be absolutely sure that the probe or nozzle was exactly where I thought it should be.
Could you please please explain why when I do a standard touch screen level with a block 5mm high on point X91 Y170 and I later return the nozzle to that point, G0 X91 Y170 Z0, the nozzle is only 3.2mm above the bed when it should be 5.1mm (5mm block + 0.1mm levelling clearance)
If I move the nozzle to point G0 X72 Y160 Z0 it is 5.1mm from the bed when it should be a lot less.
On a bed that has one single point 5mm above the rest of the bed, there should only be one point that is 5mm high and it’s in the wrong place. I dont think it’s a coincidence that it is exactly X-19 and Y-10 mm away from where it should be. There is something wrong.

This is using standard touch screen leveling and G0 move commands.

If you can look at that and let me have an explanation, I will try and work through the code and so if I can understand what is going on.

Thanks for confirming the answer to question 1 is yes.

ldp7, 2021-06-03:
I will please @scotthsl and @parachvte to take care of this issue. Thank you for your continuous support to Snapmaker company.

scotthsl, 2021-06-03:
Hi @stewartwlewis,
We will do some experiments and then analyze where the differences are and try to give a clear explanation.
Thanks for your concern!

stewartwlewis, 2021-06-03:
Thansk scotthsl

This does not cause me a great issue as I print on glass and my bed is perfectly flat (even if not level). I discovered the issue after moving to an IR sensor and wanting to check accuracy.  I started measuring exactly what was happening and found this anomaly.
Many people have trouble leveling and there is a long running thread here where @tone developed a way of "correcting" the leveling matrix using a micrometer.
https://forum.snapmaker.com/t/sm2-bed-leveling-101/6146?u=stewl
In practice, many people found this  produced a more accurate level and had/have great success with it.  If you look at the process however it actually changes the leveling positions from G1029 locations to M42 locations and in that way gets a much better result.  I dont think there is much wrong with the original probes accuracy.
What I am trying to say is it's not just me!
If you are printing on a bed that is not flat this issue could well result in the nozzle digging into the bed surface as it moves across to start a print - something we have seen so many comments about in this forum.

Kaelum, 2021-06-03:
If I have some spare time, I can take a look into this.  No promises, as I spend at least 40 hours a week working on global systems, but I will say that I have not observed the issue as presented.  With that being said, I need to know the exact version of Marlin that this code branched from.  If you branched from multiple versions, I probably won't be able to help, as I would need to take an extremely deep dive into the code, and I definitely do not have the time to do that.

Also, what is the difference between this **[Snapmaker2-Controller](https://github.com/Snapmaker/Snapmaker2-Controller)** project and the **[Snapmaker2-Modules](https://github.com/Snapmaker/Snapmaker2-Modules)** project?  It seems counter intuitive to have 2 different projects for seemingly the same purpose.

stewartwlewis, 2021-06-09:
Hi @scotthsl
Have you managed to do the test?  Did you get the same result as me?  Have you looked into this any further?
Thanks

scotthsl, 2021-06-09:
Hi @stewartwlewis ,
Sorry for late reply!

Yes, we got same result as you. And we have found the root cause for this issue.

As mentioned by @ldp7, (actually response from me), we put the coordinates with workspace  in the grid array which cause G42 use non-RAW coordinates.

During the printing process, when compensating for the Z-axis height, marlin also uses the grid array to identify which grid of the leveling matrix the XY coordinates of target location are in. The Z values of the four vertices of that grid is then used to compensate for the Z-axis height at target location.

In the above process, the XY coordinates of target location are RAW coordinates, but using RAW coordinates to find the corresponding grid in the grid array of non-RAW coordinates is exactly the step that causes the problem.

We will fix this bug as soon as possible: we plan to change the coordinate in the grid array to RAW coordinate and remove the step in the calibration process that converts non-RAW coordinate to RAW coordinate, in order to adapt marlin's native leveling-related Gcode.

Finally, thank you very much for your persistence in pointing out our problem and making us discover this potential bug.

Best Regards
Scott

stewartwlewis, 2021-06-10:
Hi Scott
Pleased you found the issue and are fixing it. 
Will the fix come as a as a separate thread of just in a new firmware release?
I dont understand exactly what you intend to do but I hope you are going to align the Snapmaker leveling routine to the Native Marlin code as in G29 and G42 as this will hopefully avoid other problems.  The reason I say this is the issue we have found in levelling also appears in the results from the G30 command.  Can you please have a read of the thread on the forum linked below.  I would hope that if you align your code, G30 will also start to work correctly.  Are you OK to look at this with the changes you are making to the levelling process or would it be better for me to start another issue?
Thanks

Stewart
https://forum.snapmaker.com/t/potential-issue-with-bed-levelling-process/18649/30?u=stewl

scotthsl, 2021-06-11:
Hi @stewartwlewis,
Thanks for your quick reply!

We have found the issue, and what we are about to do is to fix the problem you mentioned: align the Snapmaker leveling routine to the Native Marlin code.

We plan to fix this issue next week, then release a new firmware after test team confirming it is ready.

Best Regards
Scott

cmprmsd, 2021-07-07:
Has this patch been rolled out?

stewartwlewis, 2021-07-07:
Not yet though it will be in the next firmware update as it’s in the main branch now. 
Version V4.2.3 was committed in GitHub about 5 days ago. It took about 10 days for a new release after a GitHub version commit last time. 
Though I don’t know when the next firmware version will be released.

easycheese, 2021-07-18:
FYI, new firmware was just released with the fix.

stewartwlewis, 2021-07-18:
Thanks, I hope it makes a difference and people find levelling improved, I have been running the update for a while and am getting a good level across the whole bed for the first time.

closed 2021-08-13 by xhycode