Snapmaker/Snapmaker2-Controller #114: Feature Request: The ability to control flow rate

Bpositivedotdk, 2021-05-21:
I think the ability to control flow rate directly on the machine is very important. 
Various materials need a different flow and this is annoying when you need to cancel the job and make a new g-code to change it.

Would it be possible to implement on the machine control?

Bpositivedotdk, 2021-05-21:
Also, it would be awesome if you would have the ability to have material settings (flow, temp, retraction) on the machine instead of on the gcode. On my ultimaker I can store the various materials and this works perfectly.

paauserrat, 2021-05-28:
Your suggestions are amazing! I also want this to be implemented

sumibin, 2021-08-13:
Since Marlin's flow control is achieved by controlling the distance of the E-axis, real-time flow control is equivalent to regenerating a G-code.

`// Flow 200%
G1 F3600 E0
G1 F1080 X132.9 Y143.09 E0.04208
G1 X133.26 Y142.869 E0.07721
G1 X148.485 Y134.842 E1.50835
G1 X148.949 Y134.643 E1.55033
G1 X149.439 Y134.519 E1.59235
// Flow 100%
G1 F3600 E0
G1 F1080 X132.9 Y143.09 E0.02104
G1 X133.26 Y142.869 E0.0386
G1 X148.485 Y134.842 E0.75417
G1 X148.949 Y134.643 E0.77516
G1 X149.439 Y134.519 E0.79618`

brent113, 2021-08-13:
Is this not already functional in the controller with [M221](https://marlinfw.org/docs/gcode/M221.html)? A UI element just needs to be added to the touchscreen to make use of this.

Learoy1, 2022-06-22:
I asked for this basic function back when everyone first started receiving their machines. I can't believe it's still not implemented. If I can change flow while setting up the print and I can just have one G-CODE saved on the machine for each model rather than have a load of gcode files for the same model if I want to use different filament colours or brands. If I see underextrusion or overexteudion during a print I should be able to adjust flow on the machine as it prints to prevent waste and failures. It's incredibly frustrating that this expensive machine doesn't have functions that less expensive much more reasonably priced machines have. 
Please implement this asap.