Snapmaker/Snapmaker2-Controller #119: Controller Not Respecting "Build Plate Temperature Initial Layer"

iamrecursion, 2021-06-05:
When slicing to g-code both in Cura and Luban, the snapmaker controller fails to respect the temperature set for the heated bed temperature for the initial layer.

Tracyxia-tech, 2021-06-08:
Hi friend,
The default initial heated bed temp is 70℃.  During the first layer printing, what does the actual heated bed temp show?
[image]

iamrecursion, 2021-06-08:
50 C

Snapmaker-support, 2021-06-09:
When it prints the first layer, it shows 70℃ and when it prints the rest layers, the temp is 50℃. Can you check when the machine is printing the first layer, what is the temp?
Also, please check the settings in Luban to check what the initial layer temp is. Thanks

iamrecursion, 2021-06-10:
When it prints the first layer for me it's showing 50C. The settings in both Cura and Luban have 70C set as the initial layer temperature.

scyto, 2021-06-28:
For me it shows 70c.

iamrecursion, 2021-07-04:
This is odd. Any ideas how I might debug it?

scyto, 2021-07-19:
> This is odd. Any ideas how I might debug it?

- look at the gcode file and make sure it is really doing what you expect it to
- try different print paths - for example try having luban drive the print job using serial, do same over wifi, send file from luban to touchscreen *before* you try things like octoprint or other ways to send the gcode.

Kaelum, 2021-07-25:
@iamrecursion if you're using the original configuration files released by **Snapmaker** for **Cura**, they have many errors in them.  The temperatures are hard coded instead of using the values set in **Cura**, which is what you are seeing.  You should delete the **Snapmaker** machine configuration(s) and profiles that you currently have, upgrade to the latest version of **Cura**, and use one of the machine configurations that are now provided in **Cura** for the **Snapmaker 2**.

xhycode, 2021-12-14:
I will close this issue, if you have the same problem you can open it again.

closed 2021-12-14 by xhycode