Snapmaker/Snapmaker2-Controller #120: Big One! The initial calibration fails with a headbang!

CptHolzschnauz, 2021-06-07:
Hi there!

I just unpacked my brandnew A250 and went trough the initial calibration process.
When the head goes to Point 3 (Right corner up) , the lasersensor measure out of bounds and press the nozzle deep into the bed! After moving the black cover 2 cm to the top everything is fine!

**You have to change the initial procedure urgently!**

**To everyone with a new A250 (i can only tell for this machine): 

Move the plastic cover 2 cm off (away from you when sitting in front of the screen) before do the initial setup!**

Sorry for not checking if this issue is already known but this issue just ruined my black cover and hopefully not the whole machine... gdit 

May this post prevent others from damage.

closed 2021-06-08 by CptHolzschnauz

xhycode, 2021-06-08:
Hi friend,
Thank you for your feedback. We feel very sorry that there is something wrong with the machine when you first use it.

You mentioned moving the black cover 2cm to the top. Can you show me a picture of which black cover you refer to?

Thank you again and we will do the best to provide a better product and service for our customers.

CptHolzschnauz, 2021-06-08:
Hi there

It was my fault…I assembled to whole top one screw off site. Sorry for the wrong alarm, I closed that issue some hours ago

Regards

R