Snapmaker/Snapmaker2-Controller #122: Feature Request: Enclosure should start up with lights on by default

Galindorf, 2021-06-18:
# 🐞 feature request

### Is your feature request related to a problem? Please describe.

On startup, enclosure lights are automatically off, they have to be turned on manually in an awkward fashion on a separate screen of the touch pad. 
### Describe the solution you'd like?
When the Snapmaker with enclosure starts up, please have the lights be "on" by default.

iamrecursion, 2021-06-18:
Duplicate of #65.

Snapmaker-support, 2021-08-11:
Hi friend,
The coming new firmware that will support the feature is scheduled to be released this month.

scyto, 2021-09-17:
i can confirm this is fixed, yay! thanks snapmaker. time to close the issue :-)

xhycode, 2022-01-11:
This issue has been supported in [V1.13.2](https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/17) and later, please use the latest firmware.

closed 2022-01-11 by xhycode