Snapmaker/Snapmaker2-Controller #127: Feature Request: M601 Pause

Matchlighter, 2021-07-17:
Similar to M600, except it doesn't trigger a filament change. I'd love to see support for M601 so that I can pause prints to, eg, embed a magnet inside of a part. It would be really nice not to have to "swap" filament (like when using M600 to achieve this).

artemician, 2021-07-17:
This would certainly be useful to be able to manually inject M601 at desired points.

xhycode, 2021-12-13:
Hello @Matchlighter  @artemician 
The M75 suspend command is now supported in the just released [firmware](https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/17). You can manually insert the pause into gcode
Hope it will be a good experience for you!

brent113, 2021-12-13:
> Hello @Matchlighter @artemician The M75 suspend command is now supported in the just released [firmware](https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/18). 

?? https://marlinfw.org/docs/gcode/M075.html

Was this supposed to be M0?

xhycode, 2021-12-14:
Sorry for the confusion and for telling you the wrong command, the new support is for [M76](https://marlinfw.org/docs/gcode/M076.html) as a pause command instead of M75.
Because the SM2 firmware does not support control of print time, we use this command directly as a pause function.
Another reason for your confusion is that SM2 Gcode documentation has not yet been updated. We will try to improve it.
Again, sorry for the confusion!

xhycode, 2021-12-14:
For the pause command perhaps [M25](https://marlinfw.org/docs/gcode/M025.html) is a little better, and the next modification of M25 supports on the same function as the current M76. But will still retain M76, I hope this can solve your confusion.
I will close this issue, if you have the same problem you can open it again

closed 2021-12-14 by xhycode