Snapmaker/Snapmaker2-Controller #128: ISSUE: Touschreen Calibration Data invalid after changing XYZ esteps with M92

scyto, 2021-07-19:
**Issue**
After changing XYZ esteps with M92 if one performs a touchscreen calibration the resulting bilinear and catmull meshes are invalid and (in my case) placed the nozzle below the bed height destroying my build plate on one side.
 
**Repro**

1. power on A350
2. check calibration mesh is as expected using M420 V
3. change esteps values for XYZ - for example M92 X401.81 Y400.6 Z393.51
4. now perform 5x5 calibration from the touchscreen
5. now look at M420 V values

**Expected** Behaviour
See roughly same values for M420 V

**Actual** results
See incorrect values, which if used may place the nozzle below the build surface.

**More** data
My mesh values at step 2 above
```
M420 V
others < Bilinear Leveling Grid:
others <       0      1      2      3      4
others <  0 +6.033 +6.481 +6.846 +7.128 +7.364
others <  1 +6.395 +6.847 +7.131 +7.360 +7.485
others <  2 +6.561 +6.971 +7.320 +7.499 +7.584
others <  3 +6.635 +7.211 +7.264 +7.360 +7.525
others <  4 +6.539 +7.115 +7.016 +7.196 +7.431
```

My mesh values at step 5 above:
```
> M420 V
M420 V
others < Bilinear Leveling Grid:
others <       0      1      2      3      4
others <  0 +0.488 +1.001 +1.381 +1.666 +1.848
others <  1 +0.925 +1.407 +1.712 +1.934 +2.018
others <  2 +1.097 +1.539 +1.900 +2.089 +2.140
others <  3 +1.136 +1.768 +1.826 +1.926 +2.061
others <  4 +1.015 +1.620 +1.530 +1.735 +1.921
```

scyto, 2021-07-19:
Either we should be blocked from changing M92 completely. (we note custom M92 values are thrown away during power cycle)
Or touchscreen should not do this after changer M92 esteps.

I have not yet tested if using manual marlin codes for auto calibration change this - we note as a community that the touchscreen calibration is NOT using standard marlin codes to do the calibration, it behaves differently.

scyto, 2021-07-21:
(i know snapmaker don't support editing M92, but many guides include it as standard calibration procedure, this IMO requires more than just being documented)

scotthsl, 2021-07-25:
Hi @scyto ,
Sorry for late reply and thank you very much for your feedback.

 In snapmaker's firmware, currently it is not supported to save the lead configured with M92  , because the linear modules we produce later will use another lead configuration: 20mm for XY and 8mm for Z-axis (the earliest batch of linear modules produced had 8mm lead for XYZ). So the controller must automatically detect the lead of plugged XY axis, and then configure it to the planner's setting.

We are now aware of the problem you mentioned, which does cause the problem of mesh grid deviation, and we are currently thinking of a more reasonable solution to solve this problem!

Best Regards
Scott

scyto, 2021-07-27:
Hi Scott,

Thanks glad you are thinking about it.

Personally i found M92 to be key in making a 20mm XYZ calibration cube to be 20.05 x 20.05 x 19.05.  It would be good if the firmware could detect the modules and just interpret M92 correctly.  Adjusting flow is not a good solution to making a cube truly a cube.

But yes making sure folks don't break their machine is key - this destroyed my build plate :-(

xhycode, 2021-12-13:
Hi scyto
The M92 parameter saving function is supported in the just released firmware.
Sorry for the late support of this issuse
You can download the latest [firmware](https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/17) here

closed 2021-12-14 by xhycode

scyto, 2021-12-14:
Thanks!

Closed #128<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FSnapmaker%2FSnapmaker2-Controller%2Fissues%2F128&data=04%7C01%7C%7C1664b4bd26c946e0b47308d9beb44098%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637750503653420722%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=s4%2F799DWftTh2tTE8Uu87u8cntNY9jusUkxpjNB0S1Y%3D&reserved=0>.