Snapmaker/Snapmaker2-Controller #135: Feature Request: push notification

baukeplugge, 2021-08-30:
I would be nice to add push notifications to the Snapmaker .

So it would send push notifications for the following:
- job complete
- filament out
- any error occured
this can first be facilitated by for example pushover. Then you can use their services.
in a later stage a snapmaker app would be better to get this facilitated. then you also check the current % complete and maybe more.

sumibin, 2021-10-13:
This is a great idea.We are already considering related functions. But this is a long-term matter, and it still takes a long time.

mr-manuel, 2022-01-08:
I think this is a duplicate to https://github.com/Snapmaker/Snapmaker2-Controller/issues/38