Snapmaker/Snapmaker2-Controller #136: Z Offset not resetting to 0 (zero) prior to leveling

Kaelum, 2021-09-02:
I noticed this a few releases ago, so I'm not exactly sure which release it began with, but the **Z Offset** that is stored in the **Touch Pad** is no longer being reset to 0 (zero) prior to any bed leveling procedure.  This is having an accumulated affect that makes it impossible to level the machine w/o taking some steps prior to performing any bed level.  The work around is:

1. Start a print.
1. When UI allows you to, go to the **Z Offset** adjustment page.
1. Set the **Z Offset** to 0 (zero).
1. Cancel the print.
1. Power cycle the machine.

I haven't looked at the code to figure out why it is working this way, but the **Z Offset** in the **Controller** does not match the **Z Offset** in the **Touch Pad**, which is probably part of the issue.  Shouldn't the **Z Offset** that is shown in the **Touch Pad**, be the actual value that is in the **Controller**?  And when you change the **Z Offset** on the **Touch Pad** (while in a print), shouldn't it update the **Controller**?

xhycode, 2021-09-16:
Hi  Kaelum.
Sorry for the late reply.
Z offset is stored and this value will be calibrated once at home and will not add up. This is done so that after a reboot the print will remain the same as the last print, without the need to adjust the z offset again.
The values shown on the screen are synchronized with the controller, so I'm sorry that the results you describe are not reproduced.
Do you mean that each time you start printing again, the z offset distance is fixed?

Kaelum, 2021-09-16:
@xhycode I haven't tried firmware 1.13.2 yet, but on 1.13.1, the Z Offset number is not reset during a bed level, and the new level uses the previous Z Offset number as the new 0 (zero) starting position.  After you do this a few times with a negative Z Offset, your Z Offset maxes out at -0.40mm and you can no longer move the nozzle closer to the bed.  The work-around to reset the Z Offset to 0 (zero) is shown in the steps above.  That is the only way that I have been able to level the bed.

Kaelum, 2021-11-21:
@xhycode is this being addressed?  It has been almost 3 months since I reported this, and the Z Offset is still not being reset to 0 (zero) when a bed leveling (calibration) is initiated.  It is accumulating ("add up") with each and every bed level (calibration).

xhycode, 2021-11-25:
I'm sorry I can't reproduce the problem you described, if your problem still exists, you can contact our technical support at support@snapmaker.com to solve it

Kaelum, 2021-11-25:
@xhycode How to reproduce the issue:

1. Autolevel the Snapmaker where you end with a negative Z Offset and save the calibration.
2. Repeat step #1 until you can no longer lower the nozzle due to the -0.40 limit.

If the Z Offset after your first autolevel is -0.15, the second will start at -0.15 and you'll end up with a -0.30 Z Offset.  On the third attempt, after the autolevel the Z Offset will show -0.30, you won't be able to go below -0.40 due to the limit, and the nozzle will not be touching the bed.